OctoPrint/OctoPrint #4212: Printing after SD print does not work if already printed from Octoprint

computerjoe314, 2021-08-13:
### The problem

If after running an SD card print from the printer LCD, I  go and try to print from Octoprint, it just reruns the print from the SD card and does not do the requested print. 

However, this only occurs when Octoprint had previously printed something. If I restart both Octoprint and the Printer, then run the SD print, followed by an Octoprint print I have no problems. But, if I had previously printed an Octoprint print, then  a SD print, and then tried to print from Octoprint again, the printer would begin the SD print again instead of doing the selected print from Octoprint. 

However, if I click the "refresh" button on the file list before starting the Octoprint print the issue does not appear to happen.

 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

 1.6.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Creality Ender 3 Pro (32 Bit Mainboard), Marlin 2.0.9.1 (Self Compiled)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Logs:
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6983875/serial.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6983876/octoprint.log)

G-Code files: (Added a .txt so github would accept it)
On the SD card: 
[CHEP_M0_bed_level.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6983900/CHEP_M0_bed_level.gcode.txt)
A print I was trying to do:  
[xyzCalibration_cube_0.2mm_PLA_ENDER3_31m.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6983910/xyzCalibration_cube_0.2mm_PLA_ENDER3_31m.gcode.txt)
Sysinfo Bundle: 
[octoprint-systeminfo-20210813123744.zip](https://github.com/OctoPrint/OctoPrint/files/6983946/octoprint-systeminfo-20210813123744.zip)

jneilliii, 2021-08-14:
I'll test one more time tomorrow, but with OctoPrint 1.7.0c2 I was unable to reproduce this issue, so it's either fixed in the current rc release or an issue specific to your setup. To verify this is the process I followed: 

1. Print "file 1" from OctoPrint's local storage using web UI, result "file 1" printed.
2. Print "file 2" from sd card using printer's LCD (Touch Screen), result "file 2" printed.
3. Print "file 3" from Octoprint's local storage using web UI, result "file 3" printed.

jneilliii, 2021-11-11:
After the discussion on Discord, I think this is reproducible if you try to print the same file from the OP side. Here's what I did. I uploaded file_1 to local storage, file_2 to sd card. The files are simply holding M117 messages for which file it is. 

[file_1.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/7516954/file_1.gcode.txt)
[file_2.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/7516955/file_2.gcode.txt)

Process to reproduce...

1. print `file_1.gcode` from OctoPrint, results in `file 1` M117 message.
2. Print `file_2.gcode` from printer interface, results in `file 2` M117 message.
3. Print `file_1.gcode` from OctoPrint., results in `file 2` M117 message.

jneilliii, 2021-11-11:
One thing to note, printing all from OctoPrint interface changes the file that is printed as expected.

foosel, 2021-11-11:
I think I've got a repro, but I'm not sure if I'm seeing the same thing as y'all. The issue seems to be that while the printer state updates and reflects file 2, the file list doesn't and still shows file 1 as current.

Printing file 1 again with the little printer symbol works as expected, however just hitting print again (file 1 is selected after all according to the file list) reprints file 2. 

Test approach:

1. Connect to virtual printer
2. Upload file_1.gcode only to OctoPrint.
3. Upload file_2.gcode to OctoPrint & SD
4. Hit select & print for file_1.gcode
5. In the terminal, select and print file_2.gco(de) via `!!DEBUG:select_sd file_2.gco` and then `!!DEBUG:start_sd file_2.gco` (the virtual printer otherwise won't signal it actually selected the file)
6. Now file_2.gco (SD) shows as selected in the state, but file_1.gcode is still active in the file list.

Hitting print & select on file_1.gcode does the right thing. Hitting just print reprints file 2, because even though the file list shows file 1 as selected, it actually isn't.

Is this what y'all are seeing as well?

jneilliii, 2021-11-12:
I don't know if the virtual printer reacts differently or not, but during my testing I was using the select and print button when printing file_1.gcode before and after printing the file from the printer's sd card from the LCD.

jneilliii, 2021-11-12:
You are correct though @foosel, the file in the file list is still the previously printed file even though the state panel shows the SD Card file after it was printed. So then when using select and print afterwards, 

[image]

I'm assuming the click event function thinks it's already selected and just sends the start print command rather than re-selecting the already selected file? I would think there would be two possible fixes for this. Either make sure the file list selected file is in sync with the state panel or remove the if block from here?

https://github.com/OctoPrint/OctoPrint/blob/3c094f55852f322e22f8446b61caebee6f30ddc8/src/octoprint/static/js/app/viewmodels/files.js#L653-L681

jneilliii, 2021-11-12:
Removing the if block would probably be the quick fix, but updating the selected file in the list when printed from LCD seems like it would be the more proper fix. I guess the question would be do you subscribe to changes to the state panel, or just update it somewhere else on print start event?

foosel, 2021-11-15:
The file list definitely was buggy there, this should now be fixed by the above patch, and consequently also this bug. I'll take another look at the aforemention `if` though, that might be a bit TOO much of a shortcut.

commit a2e92fb4 (2021-11-15): 🐛 Fix file list selection

It wasn't updating the currently selected file correctly,
leading to issues such as #4212

Closes #4212
closed 2022-05-17 by foosel