OctoPrint/OctoPrint #4213: Auto detection of raspberry pi Camera v2

nmadura, 2021-08-14:
### The problem

### Issue
Brand new installation on new hardware. Out of the box the Raspberry Pi 2 Camera v2 (noIR version) did not work with a new Raspberry Pi 4 Model B.

There is a very simple solution (see below). The bug is that there is no way to configure from the web interface, or no assistance by the interface is offered.

### **The fix**
in `/boot/octopi.txt` I had to change the line 
`camera="auto"`
to
`camera="raspi"`

### **The backstory**
After about 2 hours troubleshooting I found that `mjpg_streamer` was running with these options `/mjpg_streamer -o output_http.so -w ./www-octopi -n --listen 127.0.0.1 -i input_uvc.so -r 640x480 -f 10 -d /dev/video0`  

Apparently `input_uvc.so` conflicts with `input_raspicam.so`
so attempting to use the builtin rasp pi utilities such as 
`raspistill -o image.jpg`
emits the following error:
`mmal_vc_component_enable: failed to enable component: ENOSPC`

which leads you down a rabbit hole ...

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.3 Safari/605.1.15
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 8134148096
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.4
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Prusa-Firmware 3.10.0 based on Marlin
systeminfo.generator: systemapi

[image]

kennethjiang, 2021-08-14:
@nmadura If you have a Pi Cam and `mjpeg_streamer` is running with `input_uvc.so`, you may want to open an issue in https://github.com/guysoft/OctoPi/.

Or better yet, you can try to dive deeper why it is the case. I believe the relevant code is at: https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/root/bin/webcamd#L237

cp2004, 2021-08-14:
This is not an OctoPrint issue, as OctoPrint doesn't deal with the camera streaming, it just consumes a stream provided by a 3rd party.

All of these options are widely documented on https://community.octoprint.org, and (if I remember right) there's links to those pages in /boot/octopi.txt.

cp2004, 2021-08-14:
Raspberry Pi cameras can also show up (and work correctly) when running as a USB device, recent Raspberry Pi firmware/v4l2 versions allow this to work. Usually only one device is allowed to use the camera, which is why raspistill doesn't work.

closed 2021-08-14 by cp2004