OctoPrint/OctoPrint #4214: 1.7.0rc1/rc2:breaks plugin "octoprint-customControl"

cmock, 2021-08-15:
### The problem

Both with 1.7.0rc1 and rc2, the plugin "customControl" (https://github.com/Salandora/octoprint-customControl) is broken; the custom controls that were previously available are gone from OctoPrint's "Control" tab, and the plugin configuration page is empty (see screenshot).

It did work fine up to 1.6.1.

I also opened an issue in the plugin's repo (https://github.com/Salandora/octoprint-customControl/issues/40), but I guess at least a release note mentioning this might be helpful.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

The issue affects a plugin

### Version of OctoPrint

1.7.0rc2

### Operating system running OctoPrint

OctoPi 0.17

### Printer model & used firmware incl. version

Kossel Mini running Klipper

### Browser and version of browser, operating system running browser

Firefox 78.12.0esr (64-bit), Debian Buster

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[octoprint-systeminfo-20210815173146.zip](https://github.com/OctoPrint/OctoPrint/files/6988398/octoprint-systeminfo-20210815173146.zip)

cp2004, 2021-08-15:
Would be interesting to see the browser JS console log - it may not be a *fixable* (from OctoPrint's side) issue but it would be good to know what the problem is.

cp2004, 2021-08-29:
Since the browser console log was not provided, there is no point in keeping this issue open. It's listed under 3rd party issues on the feedback ticket.

closed 2021-08-29 by cp2004