OctoPrint/OctoPrint #4215: ugly to failed prints with prusa mk3s (from usb)

Rayn2k18, 2021-08-16:
### The problem

Problem : prints are ugly and more likely to fail if printed through usb (exact same gcode printed from sd card are perfect)
What i saw : 
- from sd card (even while octoprint is connected and monitoring) : print is normal
- from "usb" (octoprint streaming gcode) : the nozzle has a tendancy to "dive" into printed part and extrude (doesn't bring up nozzle and sometimes make it go down).
- CPU doesn't seem to go high (1200 MHz @40-50% load max, on raspberry pi 3B+ , gpu_mem = 16)
- no pause / freeze in print, temperatures (bed, nozzle, raspi cpu) mostly constant

What i tried : 
- updated octoprint (with 1.7.0 rc1 & rc2, because i had the problem before)
- run octoprint in safe mode : same pb
- edit and compare serial.log and gcode : they were nearly the same (especially G1 Z codes, followed by "ok" message)
- reduced amount of gcode in superslicer 2.3.56 (by raising "min_length" parameter)

[gcode - photo.zip](https://github.com/OctoPrint/OctoPrint/files/6991081/gcode.-.photo.zip)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6990998/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6991088/serial.log)
[octoprint-systeminfo-20210816103011.zip](https://github.com/OctoPrint/OctoPrint/files/6990996/octoprint-systeminfo-20210816103011.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.7.0rc2 (Python 3.7.3)

### Operating system running OctoPrint

Raspbian GNU/Linux 10 (buster)

### Printer model & used firmware incl. version

Original Prusa i3 MK3s + mmu2s (Firmware 3.10.0)

### Browser and version of browser, operating system running browser

Firefox 91.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Rayn2k18, 2021-08-16:
[gcode - photo.zip](https://github.com/OctoPrint/OctoPrint/files/6991081/gcode.-.photo.zip)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6990998/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6991088/serial.log)
[octoprint-systeminfo-20210816103011.zip](https://github.com/OctoPrint/OctoPrint/files/6990996/octoprint-systeminfo-20210816103011.zip)

foosel, 2021-11-10:
First of all sorry for the late reaction, this got drowned. What's your resend ratio in general?

In your `serial.log` I'm seeing a bunch of delays whenever the MMU appears to be involved at the start, and also an `unknown command` error whenever there's an `M486` in the gcode stream (with a mangled line number however that has `N` turned into `$`, so something's fishy on firmware side here in general), e.g. 

```
2021-08-16 09:27:46,878 - Send: N1606 M486 S-1*119
2021-08-16 09:27:46,885 - Recv: Unknown M code: $1606 M486 S-1
2021-08-16 09:27:46,889 - Recv: ok
2021-08-16 09:27:46,911 - Send: N1607 M486 S0*90
2021-08-16 09:27:46,922 - Recv: Unknown M code: $1607 M486 S0
```

I would suggest to rule out that the MMU is the cause for the issues seen here. I'm not suggesting that the solution will be to not run OctoPrint with it, but I'd like to reduce the complexity of the issue at hand by ruling out one possible source of delays on the printer's side. Looking at the log I don't see anything that looks like excessive delays in sending on OctoPrint's end, most intervals between receiving an `ok` and the next line going out seem to stay below 10ms. 

You describe the issue as the nozzle diving into the print, and that sounds like either an error in your sliced files (z should never go down outside of z hops) or something in the firmware misunderstanding a valid command. I'd start with removing the MMU from the equation temporarily and also get to the bottom to where those M486 commands there are coming from, what they are supposed to achieve here, whether they are even supported by your firmware and finally why the line numbers on them get repeatedly turned from `N` into `$`.

Rayn2k18, 2021-11-11:
First, thanks for your response (on a pretty specific problem I admit ^^)

Since then, I almost got rid of my problem : 
- I completly reinstalled octoprint (full install of octopi on a new micro sd card, raw then with only usefull plugins installed)
        -> uglyness disappeared and nozzle stopped "diving"
     I think it had to do with some plugin that dispite being disabled, still interfered in the process, but I don't know which plugin it might be ...)

- And to regain former speed I tweaked gcode to speed up gradually between layer 3 and 8 from 100% to 160% print speed (in prusaslicer / superslicer, printer settings, custom gcode, before layer gcode) :

{if layer_num >= 3}{if layer_num <= 8}M220 S1{(layer_num-2)*10}{endif}{endif}

So to sum up, uglyness problem might had to do with a plugin (or missed system update ?) -> complete reinstall : ok
Speed problem : got around by increasing speed througth gcode (usb print speed @160% is now almost the same as SD print speed, weird but it works)

Thanks again

foosel, 2021-11-11:
In that case, I'll take the liberty of closing this since it sounds like it got resolved, thanks for reporting back 👍

closed 2021-11-11 by foosel