OctoPrint/OctoPrint #4218: [Request] Disable mouse scrolling when panning on the gcode canvas

MarcelloTheArcane, 2021-08-20:
### Is your feature request related to a problem? Please describe.

When I'm zoomed in close to the gcode canvas, I often use the middle mouse button (mouse wheel) to pan around, as force of habit from various CAD programs.

This has the effect of showing the mouse wheel scroll feature on the whole page, which means it scrolls up and down when panning.

The panning still works, but the page moves around too.

### Describe the solution you'd like

I ran the following Javascript in the console, and I think it fixed the issue:

```js
function preventDefaultOnMouseWheel (event) {
    if (event.which === 2) { // Middle mouse button
        event.preventDefault() // Don't pan!
    }
}
const gcodeCanvasElement = document.getElementById('gcode_canvas')
gcodeCanvasElement.addEventListener('mousedown', preventDefaultOnMouseWheel)

// Clean up:
// gcodeCanvasElement.addEventListener('mousedown', preventDefaultOnMouseWheel)
```

I'm not sure if there are any side effects I need to consider, or where this could be added.

MarcelloTheArcane, 2021-08-20:
Here's a video of what I mean:

https://user-images.githubusercontent.com/21159570/130263940-1c1ea3d1-10c8-49d9-8677-be4736059dd6.mp4

thelastWallE, 2021-10-31:
add done label pls

commit 14c2d49d (2021-11-01): 🐛 Gcode Viewer: Prevent middle mouse wheel from scrolling the entire page and panning the canvas (#4274)

Fixes #4218
closed 2022-05-17 by foosel