OctoPrint/OctoPrint #4220: [Request] secrets.yaml

vogler, 2021-08-31:
### Is your feature request related to a problem? Please describe.

I wanted to version-control my `~/.octoprint/config.yaml` but there's sensitive information in there.
The docs don't mention anything; searching the web I only found people's home-assistant configs.

### Describe the solution you'd like

It would be nice if the configuration would support something similar, e.g. referencing a git-ignored `secrets.yaml` with `api_key: !secret octoprint_key`.

cp2004, 2021-09-03:
> The docs don't mention anything

Is this meaning that there's something that needs to be improved in the docs, or is it just you didn't find anything about secrets being stored separately? Because that would be expected if it didn't exist. Just wanted to clarify 🙂

vogler, 2021-09-03:
Both? I don't get the question.
I assumed this feature does not exist because the docs don't mention anything.

This has an example for PyYAML: https://stackoverflow.com/questions/528281/how-can-i-include-a-yaml-file-inside-another