OctoPrint/OctoPrint #4227: [Request] Add info about case sensitivity to user credentials to login error

foosel, 2021-09-10:
As suggested in #2015:

> If there's still not a desire to fix the case sensitivity, maybe it would also help to add to the failed login error message that the user name is case sensitive.

_Originally posted by @QuinnDamerell in https://github.com/OctoPrint/OctoPrint/issues/2015#issuecomment-906750787_

jtowe1, 2021-09-16:
Would this addition to the failed login error only show if there was a username with only differing case?  Or would this be an addition that would always show on failed login?  Seems the latter is a better security option, since you're not revealing any extra info to someone trying to login.

foosel, 2021-09-16:
The latter, for previously that reason.