OctoPrint/OctoPrint #4231: Incorrect URL for 'Export' button in Plugin Manager when OctoPrint is not served under the server root

The-EG, 2021-09-11:
### The problem

Whenever OctoPrint is being served under a folder, not the root of the server, ie. `http://(ip)/ender-3-v2/`, the 'Export' button at the top of the Plugins Manager screen does not prefix the correct path.

Instead of linking to `http://(ip)/ender-3-v2/plugin/pluginmanager/export` the button instead targets `http://(ip)/plugin/pluginmanager/export`. If I manually go to `http://(ip)/ender-3-v2/plugin/pluginmanager/export` I get the expected file.

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.0rc2

### Operating system running OctoPrint

Raspberry Pi OS (buster)

### Browser and version of browser, operating system running browser

Waterfox G3.2.4.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20210911163943.zip](https://github.com/OctoPrint/OctoPrint/files/7148492/octoprint-systeminfo-20210911163943.zip)

cp2004, 2021-09-11:
Hmm, that shouldn't have been overlooked... I believe I added that button to the UI, so I'll look at sorting it

foosel, 2021-09-13:
@cp2004 I hope you don't mind that I quickly fixed this. I hope to be able to run through the update tests today and ideally also push out the new RC and that was kinda blocking it 😅

Trivial fix really, and overlooked during review, so this was my mistake as well 😊

cp2004, 2021-09-13:
All good

The-EG, 2021-09-13:
Confirmed fixed in 1.7.0rc3

commit 16a7ce0c (2021-09-13): 🐛 Fix URL for plugin list export

Needed baseurl prefix.

Closes #4231
closed 2021-10-11 by foosel