OctoPrint/OctoPrint #4232: [RC Feedback] Feedback on the 1.7.0rc3 Release Candidate

foosel, 2021-09-13:
Please provide *general* feedback on your experience with the 1.7.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

* #4214 / https://github.com/Salandora/octoprint-customControl/issues/40

### Issues needing analysis

SwagDad99, 2021-09-14:
My update to RC3 from the 1.6.1 stable failed.  It had lots of retries because of Connection broken 'Protocol error' timeout from www.pinwheels.org.  I tried twice on two different Pis

foosel, 2021-09-14:
@SwagDad99 that sounds like either network issues on your end or a Python infrastructure issue right when you attempted to upgrade, both outside of my control. Please try again and if it fails again share a system information bundle.

ChrisHeerschap, 2021-09-15:
Been using it pretty heavily on two systems - Pi4/MK3S and Pi3/Prusa Mini - and zero problems.

foosel, 2021-09-16:
Normally I'd say I'll release this early next week given the quite positive feedback and stats, but with the upcoming knee surgery appointment on Monday I'll see I release this ASAP after that. Probably not before the 27th.

Taomyn, 2021-09-17:
Updated from 1.6.1 on OctoPi 1.8.0, without any issues and everything including the many plug-ins all seem fine so far. Did a small set of test prints to my Ender 3pro and all was running well.

b-morgan, 2021-09-18:
Upgraded from rc2 to rc3 with no issues so far.

AndroidCloned, 2021-10-01:
Can't seem to get arcwelder to install https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc2.zip
`Unable to remove -Wstrict-prototypes or to add -Wno-unknown-pragmas.
2021-10-01 10:37:15,513 ! Building Extensions for PyArcWelder - extra_compile_args:['-O3', '-std=c++11', '-Wno-unknown-pragmas', '-v'] - extra_link_args:[]
2021-10-01 10:37:15,514 ! building 'PyArcWelder' extension
2021-10-01 10:37:15,515 ! creating build/temp.linux-armv6l-3.9
2021-10-01 10:37:15,517 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder
2021-10-01 10:37:15,518 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data
2021-10-01 10:37:15,519 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib
2021-10-01 10:37:15,520 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib/c
2021-10-01 10:37:15,522 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib/c/arc_welder
2021-10-01 10:37:15,523 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib/c/gcode_processor_lib
2021-10-01 10:37:15,525 ! creating build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib/c/py_arc_welder
2021-10-01 10:37:15,526 ! arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ioctoprint_arc_welder/data/lib/c/arc_welder -Ioctoprint_arc_welder/data/lib/c/gcode_processor_lib -Ioctoprint_arc_welder/data/lib/c/py_arc_welder -I/usr/include/python3.9 -c octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp -o build/temp.linux-armv6l-3.9/octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.o -O3 -std=c++11 -Wno-unknown-pragmas -v
2021-10-01 10:37:15,528 ! error: command 'arm-linux-gnueabihf-gcc' failed: No such file or directory
2021-10-01 10:37:15,529 ! ----------------------------------------
2021-10-01 10:37:15,530 ! ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-f6te94nz/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-f6te94nz/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-lbtfyx3y/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /mnt/dietpi_userdata/octoprint/.local/include/python3.9/Arc-Welder Check the logs for full command output.
2021-10-01 10:37:15,539 > Running setup.py install for Arc-Welder: finished with status 'error'`

sticilface, 2021-10-10:
Upgrading from 1.6.1 

get the following error
```
ImportError: cannot import name NullHandler
```

from command 

```
/home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-cTnAxi/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython
```

full log 

```
2021-10-10 13:17:59,287 > Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<1,>=0.57->OctoPrint>=1.3.5->OctoPrint-MQTT==0.8.10) (1.12.0)
2021-10-10 13:17:59,287 > Requirement already satisfied: backports-abc>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from tornado==5.1.1->OctoPrint>=1.3.5->OctoPrint-MQTT==0.8.10) (0.5)
2021-10-10 13:17:59,288 > Requirement already satisfied: singledispatch in /home/pi/oprint/lib/python2.7/site-packages (from tornado==5.1.1->OctoPrint>=1.3.5->OctoPrint-MQTT==0.8.10) (3.4.0.3)
2021-10-10 13:17:59,288 > Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python2.7/site-packages (from markdown<3.2,>=3.1->OctoPrint>=1.3.5->OctoPrint-MQTT==0.8.10) (41.0.1)
2021-10-10 13:17:59,289 > Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint>=1.3.5->OctoPrint-MQTT==0.8.10) (2.0)
2021-10-10 13:17:59,289 > Building wheels for collected packages: OctoPrint-MQTT
2021-10-10 13:17:59,289 > Building wheel for OctoPrint-MQTT (setup.py): started
2021-10-10 13:18:00,882 > Building wheel for OctoPrint-MQTT (setup.py): finished with status 'done'
2021-10-10 13:18:00,884 > Stored in directory: /tmp/pip-ephem-wheel-cache-SpPu9o/wheels/41/5d/c7/0caa6cd2816479371e0ff6753694f98ca4dc3ba9fc80ee7750
2021-10-10 13:18:00,884 > Successfully built OctoPrint-MQTT
2021-10-10 13:18:00,884 > Installing collected packages: OctoPrint-MQTT
2021-10-10 13:18:00,885 > Found existing installation: OctoPrint-MQTT 0.8.7
2021-10-10 13:18:00,885 > Uninstalling OctoPrint-MQTT-0.8.7:
2021-10-10 13:18:00,885 > Successfully uninstalled OctoPrint-MQTT-0.8.7
2021-10-10 13:18:00,886 > Successfully installed OctoPrint-MQTT-0.8.10
2021-10-10 13:19:38,081   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc3.zip --no-cache-dir
2021-10-10 13:19:39,643 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-10-10 13:19:39,645 > Collecting https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc3.zip
2021-10-10 13:19:40,647 > Downloading https://github.com/OctoPrint/OctoPrint/archive/1.7.0rc3.zip
2021-10-10 13:19:58,975 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (2021.2.23)
2021-10-10 13:19:58,976 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.2.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (2021.8.11)
2021-10-10 13:19:58,977 > Requirement already satisfied: OctoPrint-PiSupport>=2021.8.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (2021.8.2)
2021-10-10 13:19:58,977 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (1.1.1)
2021-10-10 13:19:58,978 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (3.1.1)
2021-10-10 13:19:58,978 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (2019.6.8)
2021-10-10 13:19:58,978 > Collecting flask<2,>=1.1.4 (from OctoPrint==1.7.0rc3)
2021-10-10 13:20:00,128 > Downloading https://files.pythonhosted.org/packages/e8/6d/994208daa354f68fd89a34a8bafbeaab26fda84e7af1e35bdaed02b667e6/Flask-1.1.4-py2.py3-none-any.whl (94kB)
2021-10-10 13:20:00,129 > Collecting Jinja2<3,>=2.11.3 (from OctoPrint==1.7.0rc3)
2021-10-10 13:20:00,130 > Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
2021-10-10 13:20:00,130 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (0.5.0)
2021-10-10 13:20:00,131 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (1.0.0)
2021-10-10 13:20:00,131 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (2.0)
2021-10-10 13:20:00,131 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (1.0.1)
2021-10-10 13:20:00,132 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (1.1.0)
2021-10-10 13:20:00,132 > Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint==1.7.0rc3) (0.1)
2021-10-10 13:20:00,132 > Collecting PyYAML<6,>=5.4.1 (from OctoPrint==1.7.0rc3)
2021-10-10 13:20:01,248 > Downloading https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz (175kB)
[... cut ...]
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 4041703424
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.1.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Prusa-Firmware 3.9.3 based on Marlin
systeminfo.generator: systemapi
```

cp2004, 2021-10-10:
Need to run `pip uninstall logging` to fix it. Explanation is below:

https://community.octoprint.org/t/my-octoprint-update-fails-with-importerror-cannot-import-name-nullhandler/34493

cp2004, 2021-10-11:
1.7.0 has been released 🎉

closed 2021-10-11 by cp2004