OctoPrint/OctoPrint #4237: [Request] Notification with link to FAQ if connection attempt fails

foosel, 2021-09-16:
### Is your feature request related to a problem? Please describe.

We see a ton of users running into issues with not being able to connect to their printer because its not even seen by the underlying OS, often because they fell victim to a power-only USB cable 🤬

### Describe the solution you'd like

OctoPrint should detect when users try to connect to a printer but there are either no serial candidates at all or all serial candidates fail (= might not be printers).

In this case the user should not only been told that the connection failed and be left alone with this situation, but also be presented with a link to an FAQ entry explaining how to debug the situation, e.g. https://community.octoprint.org/t/octoprint-cant-connect-to-my-printer/223

### Describe alternatives you've considered

Hunting down every single vendor who sells power-only cables indistinguishable from fully connected ones, but I already have enough going on.

foosel, 2021-12-01:
Implemented in 1.8.0:

  * If no ports (apart from AUTO) are available, so no serial ports have been currently detected, the connect button is disabled and a hint is displayed:

    [image]

  * If auto detection fails for whatever reason, a new notification is displayed in addition to the well-known "no more candidates to test" error state:

    [image]

foosel, 2021-12-01:
Note that ANY ports that can be selected from the drop down will count towards "more than AUTO". That includes anything injected via the `octoprint.comm.transport.serial.additional_port_names` hook, e.g. by the virtual printer.

foosel, 2021-12-02:
Old behaviour can be brought back by setting config flag `serial.ignoreEmptyPorts` (not exposed in UI) to `true`

cp2004, 2022-05-17:
1.8.0 has been released 🎉