OctoPrint/OctoPrint #4238: [BUG] Switching Release Channels of plugins do not downgrade

jneilliii, 2021-09-19:
### The problem

1. Open OctoPrint's Software update settings and switch a plugin to different release channel (ie Release Candidate).
2. Upgrade notice happens immediately (if newer version available) and click Upgrade button.
3. Decide to go back to Stable release channel and select that.
4. Version available updates but an upgrade notice is not received to install the non-matching latest available version from newly selected channel.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Because it involves plugin update checks/upgrades.

### Version of OctoPrint

1.6.1, 1.7.0rc3

### Operating system running OctoPrint

OctoPi 0.18

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/472

[image]

[octoprint-systeminfo-20210919131519.zip](https://github.com/OctoPrint/OctoPrint/files/7192561/octoprint-systeminfo-20210919131519.zip)

jneilliii, 2021-09-19:
Interestingly enough this only seems to happen with plugin, switching the release channel of OctoPrint itself does downgrade as expected.

electronicm, 2021-09-19:
I also had the problem with 1.7.0rc2.

thelastWallE, 2021-11-03:
I think i also saw some warning from "software update plugin" in the log after i uninstalled my plugin which was on the rc channel.

thelastWallE, 2021-11-10:
I will have a look how the OctoPrint release channel gets downgraded and will use probably that for the plugins.

foosel, 2021-11-24:
The problem is that for OctoPrint the version check for `is_current` switches from `python` to `python_unequal` in case of a prerelease channel being currently active. For plugins it will usually be the default of `python`, meaning if a newer or equal version to the one detected remotely is installed it won't offer to upgrade.

I guess I'll have to simply do the same for plugins.

OllisGit, 2022-01-10:
Morning,
currently I am also playing around with the release-channel feature and running into the missing downgrade functionality.
IMHO "most" of the developers use something like two or three channels: "Release", "Pre-Release" and "Development"

Switching between the channels shows always the correct latest available release in that channel. But if I already installed a "rc" or "dev"-version (with a higher number), it is not possible to "downgrade" to the latest stable release.

Example:
Installed 1.0.16.dev2
Now I want to switch back the lastest stable release 1.0.14

[image]

Maybe a "Forced Update" Button for the "Available"-Version is needed..could be a simple solution.

Just my 2 cents
Olli

foosel, 2022-01-10:
The above linked commit should actually already solve this cleanly going forward (will be part of 1.8.0)

OllisGit, 2022-01-10:
ah..I missed that link.

I used the latest op maintenance-branch (1.8.0.dev175+g53eb9bf97) and now it works as expected. Switching between each "channel-release" is now possible, great!

Thx a lot for the fast response!!!
Olli

commit 4e03fd9b (2021-11-24): 🐛 swu: Fix downgrade from RCs for 3rd party

Closes #4238
closed 2022-05-17 by foosel