OctoPrint/OctoPrint #4242: Cursor stays in password field after incorrect login attempt

reedy, 2021-10-01:
### The problem

If you login with an invalid username/password, the page refreshes, and the username and password fields are cleared. This is fine, but the cursor stays in the password box if it was there before logging in. It should move back to the username box.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

macOS 11.6, FF 92.0.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211001153534.zip](https://github.com/OctoPrint/OctoPrint/files/7267951/octoprint-systeminfo-20211001153534.zip)

Let me know if a video is useful...

thelastWallE, 2021-10-07:
I have a fix. To which branch should i create a PR? master?

cp2004, 2021-10-07:
Sorry, should have specified - maintenance.

commit a1dee07d (2021-11-01): 🐛Login: Refocus on username after failed login (#4249)

Fixes #4242
closed 2022-05-17 by foosel