OctoPrint/OctoPrint #4243: Can't upload files 15mb and above

Model505, 2021-10-01:
### The problem

As the title states, I can't upload files 15mb and above, I've tried safe mode and I've googled everything I can think of but everything either includes something my version doesn't have or is to old to be relevant

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

MacOS

### Printer model & used firmware incl. version

Ender 5

### Browser and version of browser, operating system running browser

Google

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211001194429.zip](https://github.com/OctoPrint/OctoPrint/files/7269465/octoprint-systeminfo-20211001194429.zip
[CE5_Wizard_Tray_Lg_Tx_Circle_12h_51m.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/7269486/CE5_Wizard_Tray_Lg_Tx_Circle_12h_51m.gcode.zip)
[WizardDiceRoofTwentyThreeHoursThirtyThreeMinutes.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/7269487/WizardDiceRoofTwentyThreeHoursThirtyThreeMinutes.gcode.zip)
)

cp2004, 2021-10-01:
Do you get any errors? What happens when you try and upload them? Can you post the contents of the browser console after you attempt to upload these files?

There's an option to configure the maximum size for file uploads, but the default is 1GB which should be sufficient. Could you check this is not configured in your config.yaml, located at `~/.octoprint/config.yaml`. It would be under [`server.uploads.maxSize`](https://docs.octoprint.org/en/master/configuration/config_yaml.html#server).

Otherwise, I can upload both your files to my OctoPrint instances fine, so it is likely some conflict somewhere, a setting or maybe a browser extension.

Please try and get access to the browser console - how you get there is dependent on the browser. You've said you are using 'Google' - that is a search engine found at https://google.com... so I'll assume you are using Chrome, the browser made by Google instead 😉. In that case, see [this guide I found](https://balsamiq.com/support/faqs/browserconsole/#google-chrome) that has some pictures to help find it.

Model505, 2021-10-01:
The error was Could not upload the file. Make sure that it is a readable, valid file with one of these extensions: .g, .gco, .gcode, I just updated Google Crome so let me try to upload it again.

Model505, 2021-10-01:
It didn't work, let me try to get to the browser console

Model505, 2021-10-01:
i found this 
Failed to load resource: net::ERR_CONNECTION_RESET
{"files":[{"DisplayLayerProgress":{"totalLayerCountWithoutOffset":"16"},"date":1632934689,"display":"CE5_dice2_5.gcode","gcodeAnalysis":{"dimensions":{"depth":180.0,"height":3.4,"width":127.65200000000002},"estimatedPrintTime":1330.8788352643933,"filament":{"tool0":{"length":1147.2292299999963,"volume":0.0}},"printingArea":{"maxX":137.752,"maxY":200.0,"maxZ":3.4,"minX":10.1,"minY":20.0,"minZ":0.0}},"hash":"981461795ec90621d24ade4a7633cd219e233b54","name":"CE5_dice2_5.gcode","origin":"local","path":"CE5_dice2_5.gcode","prints":{"failure":1,"last":{"date":1632938727.092394,"printTime":2440.5541559509993,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/CE5_dice2_5.gcode","resource":"http://octopi/api/files/local/CE5_dice2_5.gcode"},"size":2385471,"statistics":{"averagePrintTime":{"_default":2440.5541559509993},"lastPrintTime":{"_default":2440.5541559509993}},"type":"machinecode","typePath":["machinecode","gcode"]},{"children":[],"display":"D&D minis","name":"D-D_minis","origin":"local","path":"D-D_minis","refs":{"resource":"http://octopi/api/files/local/D-D_minis"},"size":1824284668,"type":"folder","typePath":["folder"]},{"DisplayLayerProgress":{"totalLayerCountWithoutOffset":"31"},"date":1630865669,"display":"Dragon_Scale_Test.gcode","gcodeAnalysis":{"dimensions":{"depth":180.0,"height":6.4,"width":116.539},"estimatedPrintTime":1821.3108919609094,"filament":{"tool0":{"length":1653.932500000001,"volume":0.0}},"printingArea":{"maxX":126.639,"maxY":200.0,"maxZ":6.4,"minX":10.1,"minY":20.0,"minZ":0.0}},"hash":"e0019ddce5f5c3865aed2125cb8b1836d35fe739","name":"Dragon_Scale_Test.gcode","origin":"local","path":"Dragon_Scale_Test.gcode","prints":{"failure":1,"last":{"date":1630868928.267979,"printTime":2993.363307462074,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/Dragon_Scale_Test.gcode","resource":"http://octopi/api/files/local/Dragon_Scale_Test.gcode"},"size":2214808,"statistics":{"averagePrintTime":{"_default":2993.363307462074},"lastPrintTime":{"_default":2993.363307462074}},"type":"machinecode","typePath":["machinecode","gcode"]},{"DisplayLayerProgress":{"totalLayerCountWithoutOffset":"19"},"date":1632505413,"display":"CE5_dice2_1.gcode","gcodeAnalysis":{"dimensions":{"depth":180.0,"height":4.0,"width":127.65200000000002},"estimatedPrintTime":1184.7094304992145,"filament":{"tool0":{"length":1027.291720000006,"volume":0.0}},"printingArea":{"maxX":137.752,"maxY":200.0,"maxZ":4.0,"minX":10.1,"minY":20.0,"minZ":0.0}},"hash":"cbd6a852f1e02c59b046fc6e462f01800e299c77","name":"CE5_dice2_1.gcode","origin":"local","path":"CE5_dice2_1.gcode","prints":{"failure":0,"last":{"date":1632507962.100051,"printTime":2100.484461716,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/CE5_dice2_1.gcode","resource":"http://octopi/api/files/local/CE5_dice2_1.gcode"},"size":1701883,"statistics":{"averagePrintTime":{"_default":2100.484461716},"lastPrintTime":{"_default":2100.484461716}},"type":"machinecode","typePath":["machinecode","gcode"]},{"children":[],"display":"LEGO and Hero Factory","name":"LEGO_and_Hero_Factory","origin":"local","path":"LEGO_and_Hero_Factory","refs":{"resource":"http://octopi/api/files/local/LEGO_and_Hero_Factory"},"size":37156674,"type":"folder","typePath":["folder"]},{"children":[],"display":"Battletech","name":"Battletech","origin":"local","path":"Battletech","refs":{"resource":"http://octopi/api/files/local/Battletech"},"size":7486762,"type":"folder","typePath":["folder"]},{"DisplayLayerProgress":{"totalLayerCountWithoutOffset":"76"},"date":1628966459,"display":"Test.gcode","gcodeAnalysis":{"dimensions":{"depth":180.0,"height":15.4,"width":160.109},"estimatedPrintTime":3287.989124694517,"filament":{"tool0":{"length":2404.4858600000043,"volume":0.0}},"printingArea":{"maxX":170.209,"maxY":200.0,"maxZ":15.4,"minX":10.1,"minY":20.0,"minZ":0.0}},"hash":"e6c7024a9032f497588d4cd496d220e4cc64e903","name":"Test.gcode","origin":"local","path":"Test.gcode","prints":{"failure":0,"last":{"date":1628973191.228382,"printTime":6411.27686404399,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/Test.gcode","resource":"http://octopi/api/files/local/Test.gcode"},"size":6757502,"statistics":{"averagePrintTime":{"_default":6411.27686404399},"lastPrintTime":{"_default":6411.27686404399}},"type":"machinecode","typePath":["machinecode","gcode"]},{"DisplayLayerProgress":{"totalLayerCountWithoutOffset":"16"},"date":1632508287,"display":"CE5_dice2_2.gcode","gcodeAnalysis":{"dimensions":{"depth":180.0,"height":3.4,"width":127.65200000000002},"estimatedPrintTime":1233.1055367503698,"filament":{"tool0":{"length":1050.0259699999976,"volume":0.0}},"printingArea":{"maxX":137.752,"maxY":200.0,"maxZ":3.4,"minX":10.1,"minY":20.0,"minZ":0.0}},"hash":"d195972d7cc180540ec68fe9331eb5c39de08c63","name":"CE5_dice2_2.gcode","origin":"local","path":"CE5_dice2_2.gcode","prints":{"failure":0,"last":{"date":1632510908.745564,"printTime":2498.6814645990003,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/CE5_dice2_2.gcode","resource":"http://octopi/api/files/local/CE5_dice2_2.gcode"},"size":2862438,"statistics":{"averagePrintTime":{"_default":2498.6814645990003},"lastPrintTime":{"_default":2498.6814645990003}},"type":"machinecode","typePath":["machinecode","gcode"]},{"DisplayLayerProgress":{"totalLayerCou [... cut ...] ","prints":{"failure":0,"last":{"date":1629492278.440622,"printTime":12478.841468409984,"success":true},"success":1},"refs":{"download":"http://octopi/downloads/files/local/CE5_Adrex_Horn_Mirror.gcode","resource":"http://octopi/api/files/local/CE5_Adrex_Horn_Mirror.gcode"},"size":9782008,"statistics":{"averagePrintTime":{"_default":12478.841468409984},"lastPrintTime":{"_default":12478.841468409984}},"type":"machinecode","typePath":["machinecode","gcode"]}],"free":21307527168,"total":30322888704}

Model505, 2021-10-02:
I figured out it was the browser.

closed 2021-10-02 by Model505