OctoPrint/OctoPrint #4254: Update from 1.6.1 to 1.7.0 fails on assertion error

rmm200, 2021-10-11:
Exception:Traceback (most recent call last):File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/cli/base_command.py", line 176, in mainstatus = self.run(options, args)File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/commands/install.py", line 346, in runsession=session, autobuilding=TrueFile "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_internal/wheel.py", line 886, in buildassert have_directory_for_buildAssertionErrorThe update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

Sorry about the lousy formatting...

jneilliii, 2021-10-12:
Please completely fill out a bug report, with all requested information, including the system bundle. 

https://github.com/OctoPrint/OctoPrint/issues/new?assignees=&labels=&template=bug_report.yml

ryamoo, 2021-10-12:
Looks like a good time to read the 1.7.0 release notes and upgrade to python 3 to me.

foosel, 2021-10-12:
This looks like a broken python environment, not a bug, especially considering that almost 8k instances were already updated successfully in the field according to the usage tracking. Please get in touch on the [community forums](https://community.octoprint.org) and be sure to fill out the whole reporting template there, providing all requested info, otherwise it will be next to impossible to help you.

Side note: A switch to Python 3 is indeed in order, though not the core issue here.

rmm200, 2021-10-12:
Foosel you are a savior!  Updating Pip as in the FAQ fixed it. So much for the 8,000 successful cases.
My failure was the only one I cared about.