OctoPrint/OctoPrint #4264: [Request] Have error message popup show last SEND command

ImagineerNL, 2021-10-16:
### Is your feature request related to a problem? Please describe.

When a halting error occurs, the popup shows the error message that was received from printer 
When the terminal is set to autoscroll it can happen that the original command causing the error does not show up anymore.

This makes troubleshooting difficult

### Describe the solution you'd like

It would be great if the error message popup also showed the last known SEND command.
I think this should be part of CORE as it is part of base error handling.

foosel, 2021-11-24:
Hm... The problem is that the last send command due to firmware side buffering isn't necessarily the problematic command (if there even *was* a problematic command that is, e.g. in case of sudden thermal failure). A bit of context would probably be nice though. Might make sense to attach the last 10 or so lines of the communication to the error popup as an expandable additional context and/or log them in octoprint.log.

ImagineerNL, 2021-11-24:
True, it isnt always the problematic command, but at least it is a start.
Your remark is a good one.
If you choose to go your route, then please make sure you have at least the last 'send' command and not only the last 'received' lines

To me as a non-programmer it feels easier to always have the last 'send' in a savestate, as it could be 10, but also 50 lines of statusupdates by the printer that are without any use.

Might also be an idea to make the 'terminal' window stop autoscrolling when an error state occurs.