OctoPrint/OctoPrint #4267: Special Character

PCx43, 2021-10-17:
### The problem

If there is a # in the file name of the gcode, Octo does nothing.
No deleting, no moving, nothing possible.
Just a backup without uploading and metadata, then restore and the file is gone. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.0

### Operating system running OctoPrint

0.18.0

### Printer model & used firmware incl. version

CCR10SPROV2

### Browser and version of browser, operating system running browser

all

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2021-10-17:
The issue form asked you to provide a system info bundle. That's missing. please provide one.

jcbarstow, 2021-10-21:
To be clearer, I renamed the problem files in the shell session after I found them in the file system.

commit 3bbd3f7a (2021-11-01): 🐛 URL encode all uploaded file names

Otherwise a # in the URL might be treated as a fragment and lead to
resource mismatches.

Closes #4267
closed 2021-11-01 by foosel