OctoPrint/OctoPrint #4268: cannot update version

cooktech, 2021-10-19:
### The problem

I posted this and it was removed when was advised that there was an issue with printgenius, but the problem persists without without that plugin.   Also, it does it in safe mode as I mentioned. 

ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-Cjf9Ee/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython
cwd: None
Complete output (24 lines):
Traceback (most recent call last):
File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
"__main__", fname, loader, pkg_name)
File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
exec code in run_globals
File "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
from pip._internal.main import main as _main  # isort:skip # noqa
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/main.py", line 13, in <module>
from pip._internal.cli.autocompletion import autocomplete
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 11, in <module>
from pip._internal.cli.main_parser import create_main_parser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
from pip._internal.cli import cmdoptions
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py", line 25, in <module>
from pip._internal.locations import USER_CACHE_DIR, get_src_prefix
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/locations.py", line 18, in <module>
from pip._internal.utils import appdirs
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/appdirs.py", line 16, in <module>
from pip._internal.utils.compat import WINDOWS, expanduser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/compat.py", line 17, in <module>
from pip._vendor.urllib3.util import IS_PYOPENSSL
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_vendor/urllib3/__init__.py", line 21, in <module>
from logging import NullHandler
ImportError: cannot import name NullHandler
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-Cjf9Ee/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython Check the logs for full command output.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

pctopi 0.17

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-Cjf9Ee/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython
cwd: None
Complete output (24 lines):
Traceback (most recent call last):
File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
"__main__", fname, loader, pkg_name)
File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
exec code in run_globals
File "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
from pip._internal.main import main as _main  # isort:skip # noqa
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/main.py", line 13, in <module>
from pip._internal.cli.autocompletion import autocomplete
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 11, in <module>
from pip._internal.cli.main_parser import create_main_parser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
from pip._internal.cli import cmdoptions
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py", line 25, in <module>
from pip._internal.locations import USER_CACHE_DIR, get_src_prefix
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/locations.py", line 18, in <module>
from pip._internal.utils import appdirs
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/appdirs.py", line 16, in <module>
from pip._internal.utils.compat import WINDOWS, expanduser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/compat.py", line 17, in <module>
from pip._vendor.urllib3.util import IS_PYOPENSSL
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_vendor/urllib3/__init__.py", line 21, in <module>
from logging import NullHandler
ImportError: cannot import name NullHandler
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 /home/pi/oprint/local/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-Cjf9Ee/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel Cython Check the logs for full command output.

foosel, 2021-10-19:
The problem isn't Print Time Genius, the problem is a third party dependency that got **pulled in** by an older version of Print Time Genius, this issue is covered in the FAQ including a fix and said FAQ entry can be found in an issue even pinned to this very issue tracker: #4256 

Please use the search, these kind of repeated tickets on the very same and fully solved issue not even caused by OctoPrint itself eat up a ridiculous amount of time better spent elsewhere 😕

closed 2021-10-19 by foosel

cp2004, 2021-10-19:
Even a duplicate **of your own issue** already opened

cooktech, 2021-10-19:
I am sorry - I do not normally use this process.  I will try to avoid
this.  I thought my issue was closed.

Bill Cook
CookTechnology.com
+1 786 383 2487

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
10/19/21,
09:40:14 AM