OctoPrint/OctoPrint #4271: Error during dev env setup: pip install -e '.[develop,plugins,docs]'

Jakobud, 2021-10-21:
### The problem

I am in the process of trying to setup Octoprint's dev environment in a Ubuntu 20.x Vagrant VM. When I get to this step in the setup:

`pip install -e '.[develop,plugins,docs]'`

I get the following output:

```
(venv3) root@ubuntu-focal:~/devel/OctoPrint# pip install -e '.[develop,plugins,docs]'
Obtaining file:///root/devel/OctoPrint
    ERROR: Command errored out with exit status 1:
     command: /root/devel/OctoPrint/venv3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/root/devel/OctoPrint/setup.py'"'"'; __file__='"'"'/root/devel/OctoPrint/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info
         cwd: /root/devel/OctoPrint/
    Complete output (37 lines):
    WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f09a9bc08e0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f09a9bc0490>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f09a9bc0b50>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f09a9bc07f0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f09a9bc0700>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    ERROR: Could not find a version that satisfies the requirement markdown<3.2,>=3.1 (from versions: none)
    ERROR: No matching distribution found for markdown<3.2,>=3.1
    Traceback (most recent call last):
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/root/devel/OctoPrint/venv3/bin/python', '-m', 'pip', '--disable-pip-version-check', 
'wheel', '--no-deps', '-w', '/tmp/tmpurd8q3x6', '--quiet', 'markdown<3.2,>=3.1']' returned non-zero exit status 1.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/root/devel/OctoPrint/setup.py", line 319, in <module>
        setuptools.setup(**params())
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires  
        dist.fetch_build_eggs(dist.setup_requires)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/dist.py", line 718, in fetch_build_eggs
        resolved_dists = pkg_resources.working_set.resolve(
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 780, in resolve
        dist = best[req.key] = env.best_match(
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/dist.py", line 777, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/root/devel/OctoPrint/venv3/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpurd8q3x6', '--quiet', 'markdown<3.2,>=3.1']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

It looks like something to do with pip not being able to find a valid markdown library or something... I'm not 100% sure what to do here.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

n/a

### Version of OctoPrint

n/a: latest master branch

### Operating system running OctoPrint

Ubuntu 20.04.3 LTS Vagrant VM

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Jakobud, 2021-10-21:
I think it's the same output:

```
(venv3) root@ubuntu-focal:~/devel/OctoPrint# pip install -e .[develop,plugins,docs]
Obtaining file:///root/devel/OctoPrint
    ERROR: Command errored out with exit status 1:
     command: /root/devel/OctoPrint/venv3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/root/devel/OctoPrint/setup.py'"'"'; __file__='"'"'/root/devel/OctoPrint/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info
         cwd: /root/devel/OctoPrint/
    Complete output (37 lines):
    WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f23156c18b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f23156c1460>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f23156c1b20>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f23156c17c0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f23156c16d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/markdown/
    ERROR: Could not find a version that satisfies the requirement markdown<3.2,>=3.1 (from versions: none)
    ERROR: No matching distribution found for markdown<3.2,>=3.1
    Traceback (most recent call last):
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/root/devel/OctoPrint/venv3/bin/python', '-m', 'pip', '--disable-pip-version-check', 
'wheel', '--no-deps', '-w', '/tmp/tmp_056zyr1', '--quiet', 'markdown<3.2,>=3.1']' returned non-zero exit status 1.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/root/devel/OctoPrint/setup.py", line 319, in <module>
        setuptools.setup(**params())
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires  
        dist.fetch_build_eggs(dist.setup_requires)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/dist.py", line 718, in fetch_build_eggs
        resolved_dists = pkg_resources.working_set.resolve(
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 780, in resolve
        dist = best[req.key] = env.best_match(
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/dist.py", line 777, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/root/devel/OctoPrint/venv3/lib/python3.8/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/root/devel/OctoPrint/venv3/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp_056zyr1', '--quiet', 'markdown<3.2,>=3.1']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

foosel, 2021-11-10:
Closing as this is not an OctoPrint issue.

closed 2021-11-10 by foosel