OctoPrint/OctoPrint #4292: Printer refuses to connect after setting low latency flag fails

foosel, 2021-11-01:
### Problem

Since rolling out 1.7.0, there have been a small number of people running into connection issues when setting low latency mode on the serial port fails, e.g. [on this thread](https://community.octoprint.org/t/octoprint-not-connecting-to-printer-anymore/38394) and also some comments here and there on Discord and elsewhere.

While the error resulting from failing to set the low latency flag itself is gracefully handed, apparently the involved serial ports don't even like the attempt and then no longer work correctly.

### Solution

Short term as part of the bug fix release 1.7.1: Turn requesting low latency mode *off* by default.

Longer term: Maybe close and reopen the port on mode setting failure? That would however require someone who can reproduce the issue to collaborate so possible mitigations can be tested. I currently don't have discovered any system on hand that shows the problematic behaviour and thus can only guess so far.

unixneo, 2021-11-03:
If would be good if OctoPrint would be designed  / upgraded / modified / changed to not have the kind of "we decided to add low latency code,  no matter if you need it or not" features.

For us OctoPrint was running fine.  We upgraded to 1.7.0 and it was a mistake.  Our printing (and OctoPrint standard logging) went from "well done" to a loud, "WTF?".    We ended up commenting out the "low latency" buggy code because it crashed so often (failed to connect).  Now, we see in 1.7.1 this code is back out (or optional).

May I kindly suggest for the future releases that OctoPrint refrain from adding features as defaults with a narrow view that "one size fits and is best for all", and add features with options to enable or disable them, preferably disabled by default.

For me, I thought the 1.7.0 "low latency" rollout was a disaster because it was enabled by default; there was no option to turn it off / on and it was very buggy and caused OctoPrint to crash all the time.  OBTW, we print via USB directly from the mac, so please stop assuming OctoPrint is only used via PI.   We don't need a PI when we have full design environment with 64 GB of RAM running Fusion360 / Cura, etc.  USB works well :) :)

Thank you again!

Why strong arm users to use features they do not require?    A simple "default off, enable if you like" would be a much better approach, in my view.

Thanks!

cp2004, 2021-11-03:
We had maybe 2 confirmed reports in 20 days I think that people's printers were failing to connect due to the low latency setting, out of thousands (probably hundreds of thousands) of people who upgraded to 1.7.0. 

We can't help you if you don't report anything to us.

foosel, 2021-11-03:
> Why "strong arm" users to use features they do not require? A simple "default off, enable if you like" would be a much better approach, in my view. Please.

Because this is the kind of stuff that might have a huge impact on the often complained about throughput, and most users out there - as has shown the past almost nine years of maintaining this project - don't actually look through available options, set it to default and then complain that it's not the optimal solution.

There was a several months long release candidate phase with this change in place. I was worried about exactly something like this to happen when I merged the PR bringing it in (which is also why I pushed for a config setting to support to disable it), but the almost 2k instances this was tested against during RC didn't show this issue (or if they did, the people affected never reported back). So it sounded like it was not going to cause issues. Then we got two or so reports about issues that were mitigated by switching this feature off again (which we could instruct the affected users to easily do via the config dialog - yes there was an option there) and I decided to default it to false on 1.7.1 which was primarily put out because of this.

Writing software means making decisions. I can't make everything optional, and especially with potentially quality of life improvements you *have* to make these decisions for the users because often they either don't care or are simply otherwise overwhelmed with the possible options. Everything can be configured, but finding sensible defaults is the real challenge that *has* to be navigated.

If you want to reduce the likelihood of any kind of changes to affect your workflow negatively then help test the release candidates, apparently your workflow and setup was not properly represented in the set of testers yet.

foosel, 2021-11-03:
> OBTW, we print via USB directly from the mac, so please stop assuming OctoPrint is only used via PI.

I go through a TON of pain to keep OctoPrint running cross platform, simply going with a Pi-only approach would make my life a HELL of a lot easier. I do most certainly not even remotely assume OctoPrint is only used via a Pi, even though it is in 99% of cases. So please stop making uninformed assumptions about my informed assumptions, thank you.

unixneo, 2021-11-03:
LOL when we read almost all docs, tutorials  etc on the OctoPrint website
is is all about PI PI PI.

Maybe you should learn to listen more to users than lecture to them?

You are not the only person on the planet who does software development,
believe it or not!

jneilliii, 2021-11-03:
> Maybe you should learn to listen more to users than lecture to them?
>
> You are not the only person on the planet who does software development,
believe it or not!

I feel this is uncalled for and a misdirected statement. If you aren't going to be constructive and aid in moving the project forward maybe you should look for another alternative.

foosel, 2021-11-03:
I agree with @jneilliii. Have a good day.

draeath, 2021-11-12:
Well, shoot. That was my PR and I feel crappy about it causing people problems :(

foosel, 2021-12-02:
I'm going to close this as at least the buggy situation seems to have seem fixed now with not using the low latency flag by default.

Long term I'd still like to find a printer this affected and see I can find a work around to fully recover from a failed attempt automatically so we can default to this again, but until this is the case, this will have to stay off by default 🤷‍♀️

closed 2021-12-02 by foosel