OctoPrint/OctoPrint #4294: [1.7.x] Plugin blacklist version processing

kantlivelong, 2021-11-02:
### The problem

```
2021-11-01 20:26:39,203 - octoprint.plugin.core - WARNING - Invalid version requirement >=0.1.1,<=0.2.8 for blacklist entry GcodeEditor, ignoring
2021-11-01 20:26:39,204 - octoprint.plugin.core - WARNING - Invalid version requirement >=0.1.0,<=0.1.1 for blacklist entry gcodeleveling, ignoring
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.7.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Noticed this in a users log. Seems like blacklist version processing might have a bug. Need to investigate further.

kantlivelong, 2021-11-03:
So it looks like this only occurs when the blacklist is not cached. Am able to replicate by deleting `plugin_blacklist.yaml`.

Issue is in [this](https://github.com/OctoPrint/OctoPrint/blob/d60c5a631aedf2328e4492125d9b86f595e43d00/src/octoprint/plugin/core.py#L851-L865) particular block of code.

When started w/ a blacklist in cache `plugin_blacklist` looks like this:
```
plugin_blacklist = {list: 3} ['roomtemp', ['GcodeEditor', '>=0.1.1,<=0.2.8'], ['gcodeleveling', '>=0.1.0,<=0.1.1']]
 0 = {str} 'roomtemp'
 1 = {list: 2} ['GcodeEditor', '>=0.1.1,<=0.2.8']
 2 = {list: 2} ['gcodeleveling', '>=0.1.0,<=0.1.1']
 __len__ = {int} 3
```

when not cached/deleted it's this:
```
plugin_blacklist = {list: 3} ['roomtemp', ('GcodeEditor', '>=0.1.1,<=0.2.8'), ('gcodeleveling', '>=0.1.0,<=0.1.1')]
 0 = {str} 'roomtemp'
 1 = {tuple: 2} ('GcodeEditor', '>=0.1.1,<=0.2.8')
 2 = {tuple: 2} ('gcodeleveling', '>=0.1.0,<=0.1.1')
 __len__ = {int} 3
```

Installing a blacklisted plugin such as `gcodeleveling==0.1.1` will actually be enabled for the first startup after the cached blacklist is deleted.

kantlivelong, 2021-11-03:
Created PR #4296

foosel, 2021-11-03:
I came to the same conclusion as to where the issue lies, but I don't see how there's a difference between cached and not cached behaviour here. In both cases there will be a missing package name in the current `pkg_resources.Requirement.parse` call, won't it? 🤔

foosel, 2021-11-03:
I've merged your PR after a small adjustment (as it was it wouldn't have worked), and also cherry-picked it to `staging/bugfix`. If there's a 1.7.3 (which for now I *don't* actively plan on, but it depends on what else comes) it'll get pushed out with that (hence the Milestone), otherwise it'll be 1.8.0.

Also an annoying thing that I could have spotted myself in the logs. I first thought it was not even a 1.7.x regression, but it turns out that yes, indeed, even though this blacklist parsing change was added back in April it only now made it into a stable release.

kantlivelong, 2021-11-03:
Oh interesting, it seemed to work when I tested it according to the log anyway.

What I'm also just noticing is that regardless of the validity of pluginversions in the blacklist json file the file will still cached with said values. Then on the next start the whole `pkg_resources.Requirement.parse` check is skipped because it was loaded from cache (tuple vs list check). So while the PR fixed the immediate problem it seems like there may be a bit more to it.

PR #4296 (2021-11-03): 🐛 Fix issue with processing plugin blacklist version strings on
#### What does this PR do and why is it necessary?
Fixes #4294

#### How was it tested? How can it be tested by the reviewer?
Tested w/ Python 3.8
See #4294

#### Any background context you want to provide?

#### What are the relevant tickets if any?
#4294

#### Screenshots (if appropriate)

#### Further notes
commit 34828e2f (2021-11-03): 🐛 Fix blacklist processing (#4296)

Closes #4294
commit 270cb4f9 (2021-11-03): 🐛 Fix blacklist processing (#4296)

Closes #4294
closed 2022-01-20 by foosel