OctoPrint/OctoPrint #4297: Remember timelapse setting between updates

Wintaru, 2021-11-03:
### The problem

The Timelapse setting is reset every time Octoprint updates.

1. Set setting to Off
2. Save
3. Operate Octoprint while waiting for an update
4. Update Octoprint when update is pushed
5. Observe that the setting is no longer set to off and is instead reset to "Timed".

It takes me a while to remember that this happens and my SD card is full of timelapses I don't want, this feels like something easy to do?

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.0, 1.7.2 both had this

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Qidi Tech 1

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-11-03:
Do you check 'save as default' in the timelapse options?

Wintaru, 2021-11-03:
> Do you check 'save as default' in the timelapse options?

Oh for crying out loud, it's right there in front of my face. I wondered why nobody complained about this and now I know why, thank you!

closed 2021-11-03 by Wintaru