OctoPrint/OctoPrint #4299: PrintFailed event is fired rapidly

fritzw, 2021-11-03:
### The problem

1. Use Klipper
2. Turn on the printer, but do not issue a FIRMWARE_RESET
3. Try to start a print
4. Check the terminal to see OctoPrint sending many G-Code lines and Klipper responding with the same error message each time
5. Observe that the PrintFailed event is fired many times. Probably once for each G-Code line sent (for example by using the OctoPrint-IFTTT plugin to send a push message to your phone).

Example terminal output:
```
Send: G1 E-5 F2000
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok
Send: G1 X-10 F8000
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok
Send: G1 Y200
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok
Send: M106 S0
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok
Send: G92 E0
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Requires plugins

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

octopi 0.18.0

### Printer model & used firmware incl. version

Anet A8 with Klipper (master branch on 2021-10-31 d23c1b06c6e)

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211103213920.zip](https://github.com/OctoPrint/OctoPrint/files/7470819/octoprint-systeminfo-20211103213920.zip)

[image]

foosel, 2021-11-10:
How have you configured error handling?

[image]

fritzw, 2021-11-17:
It's set to "Cancel any ongoing prints but stay connected to the printer", just as your screenshot.

NotExpectedYet, 2021-11-20:
I have "a bit" more information here for this @foosel 

It happens with cancelled prints and error's. I think it's more to do with Klipper constantly repeating the error message and OctoPrint reacting correctly too it but I don't exactly understand the specifics. 

https://github.com/OctoFarm/OctoFarm/discussions/416

fritzw, 2021-11-20:
@NotExpectedYet For me it doesn't happen when I click "Cancel" in OctoPrint, but for example when I shut the printer's power off in the middle of the print (emergency stop).

But yes, as you said Klipper repeats the error for every GCode line sent by Octoprint after the power is removed (or before homing for the first time, as stated in the original report). I'd say that this is the expected behavior: when the firmware (Klipper) receives a GCode command that it cannot handle, it should reply with an error message.

NotExpectedYet, 2021-11-20:
> @NotExpectedYet For me it doesn't happen when I click "Cancel" in OctoPrint, but for example when I shut the printer's power off in the middle of the print (emergency stop).
>
> But yes, as you said [...]

Interesting, I only really see this issue from the websocket side on OctoFarm but never really hit the situation with an emergancy stop personally. Will test more if I manage to get my printers back up this week.

foosel, 2021-11-23:
So, this is a tricky one. OctoPrint assumes a printer is still responsive to GCODE after cancelling, so it can run the cancel script. But with this specific error mode in Klipper, it isn't. Considering that prints or any other non Klipper specific commands are utterly impossible apparently in this mode, I wonder if this shouldn't actually get reported by Klipper as a fatal error, considering that the serial connection in this state is pretty much unusable for a host that expects to talk to a printer. Or is there a way to get things going again?

@KevinOConnor do you have any thoughts on this?

In any case, I've now pushed a change that at least will end the "Print Failed" storm (code wasn't checking if the print was already in the process of getting cancelled and so triggering everything with each line from the cancel gcode or that was still buffered and responded in an error from the firmware), and that will be part of 1.8.0. However, that will now mean that OctoPrint will instead get stuck in "Cancelling" because it cannot send the cancel code and actually ever complete the state transition.

I'm not sure how best to treat this error mode on my end right now, I'm leaning towards simply reading it as a fatal error and forcing a disconnect, simply because OctoPrint can't talk to something it can't speak GCODE with, and that seems to go out of the window when the MCU disconnects.

KevinOConnor, 2021-11-24:
If Klipper detects an internal error, then it'll transition into a "shutdown" state.  This is intended as a safety mode, with a priority on disabling heaters/motors and making sure they stay disabled until a user acknowledges and clears the error.

Not all commands are disabled in this state - for example, `M105` and similar status commands should still work.  Any command that requests movement (eg, `G1`) or similar would result in an error response from Klipper.  A command to clear the error state (eg, `FIRMWARE_RESTART`) would also be accepted (though we wouldn't want this type of command to be automatically sent from software as the goal of the error state is to ensure the hardware is not in a dangerous state before resuming normal operation).

I'm not sure what alternative is being proposed here.  OctoPrint can close the connection if it wishes - that doesn't have any impact on Klipper itself - but it would then require the user to reconnect to issue a `FIRMWARE_RESTART` command.

Cheers,
-Kevin

foosel, 2021-11-24:
> I'm not sure what alternative is being proposed here. OctoPrint can close the connection if it wishes - that doesn't have any impact on Klipper itself - but it would then require the user to reconnect to issue a `FIRMWARE_RESTART` command.

The problem is that OctoPrint's state handling simply doesn't currently have anything that matches "is still there but only responds to a subset of commands without an error". This is something that is Klipper exclusive so far, and staying connected in a meaningful way but blocking e.g. prints from starting or anything else that is not available in this kind of shutdown state would require adding a new state and a TON of checks against that all over the place and in third party plugins. In short, I'm not sure if simply treating this like a fatal error in other firmware, as in disconnecting with a big fat error, wouldn't be the best solution here. 

It's good to know though that apparently some commands are still possible in this mode. Can you give me some more information on possible commands in that mode? What commands exactly trigger this error and which are still allowed? Should I rather lean towards an allow or a forbid list here if I were to implement a heavily regulated "restricted" printer state to accommodate this on the host side?

KevinOConnor, 2021-11-24:
I think I'm missing the "big picture".  I've used Klipper with OctoPrint for years and haven't had an issue with the current behavior.  So, I'm not sure why any change would be needed to OctoPrint.

As long as OctoPrint recognizes an error message (`!!`) and cancels any current print (and cancels any macro or script) then everything should just work as far as I know.  Eventually the user should stop getting error reports, they can investigate, and then clear the error state.  This mode is intended as an infrequent event, and only to relay to the user that a significant problem has occurred.

I haven't done an exhaustive search, but I believe the following commands are still allowed: `M105`, `M110`, `M112`, `M115`, `RESTART`, `FIRMWARE_RESTART`, `STATUS`, `HELP`

-Kevin

foosel, 2021-11-24:
OctoPrint's cancel routine involves sending a cancel script, by default to shut off the heaters and move the print head out of the way. And in the MCU disconnect case it can't run the cancel script because the it will end in errors. It will get stuck in the Cancelling state due to this. An error by the firmware with the setting to only cancel on error (instead of outright disconnecting) will therefore cause issues in this very specific scenario here, as in, the communication will get stuck because the state change can't proceed, and the connection is now by all means pretty dead. 

Disconnecting would be easy here (and quite well supported across the board of addons). Not sending the cancel script for this specific error and just stopping the print stream would be an option, but the question is, what to do *then* because now the printer is in a semi broken state (from OctoPrint's POV), as it no longer accepts anything but a very small subset of commands (thanks for those btw!). Introducing some kind of "Restricted" state that's less severe than "Error" (which always triggers a disconnect) but far from "Operational" would be a possible solution here, but it comes with a whole rat's tail of treating that properly across the whole ecosystem.

Does this clarify the problem I'm trying to find a good solution for here?

fritzw, 2021-11-24:
In my humble opinion, a new state is not strictly necessary, and the printer could just be shown as „operational“. As long as the PrintFailed event is emitted only once for each print started. In other words, PrintFailed should only be emitted when the current state is „printing“ and the new state is „canceling“ or „error“ (but not „print successful“) and never if the current state is something else than „printing“. Errors from the cancel script and from already buffered gcode lines should then be silent, right? (I’m preemptively sorry if I misunderstood the state model, since I didn‘t look at the code, just at your explanations above).

cp2004, 2021-11-24:
What does OctoPrint need to get out of the cancelling state?

Klipper is still sending an `ok` each time, required for communication, so if OctoPrint 'tries' to send the usual gcode script, as it does now, and then gets to the end, what is it waiting for?

I think I might be with @KevinOConnor here - a shift in the current behaviour isn't necessary, just stop repeating the 'PrintFailed' event. There's a lot of people who this would impact, as they may turn their printer off using a smart plug or something, but leave the RPi on. Put a `FIRMWARE_RESTART` command in a script in the PSU control plugin and you could be automatically back up and running - disconnecting makes this much harder.

KevinOConnor, 2021-11-24:
I see.  Is there some way to detect if an error is reported during the execution of the cancel script, and then transition to the cancelled state without resending that script?  Even in the general case, if a user supplied cancel script is reporting an error, then there's little gain in re-running it.

> the question is, what to do then because now the printer is in a semi broken state (from OctoPrint's POV), as it no longer accepts anything but a very small subset of commands

FYI, the connection is only in this limited state until the user clears the error.  Once cleared, the full command set returns.  The clearing of the error isn't something that OctoPrint will typically notice (the connection is not reset nor anything similar).

It's also worth pointing out that most error reports from Klipper are not associated with a transition into a "shutdown state".  For example, if Klipper is commanded to move the toolhead to a position that is outside of its configured limits then Klipper will report a error for that command, but it will not transition to a shutdown state - typical "cancel print commands" continue to work.

-Kevin

cp2004, 2021-11-24:
I tested it quickly with the new virtual printer option - I'm not seeing this issue:
> However, that will now mean that OctoPrint will instead get stuck in "Cancelling" because it cannot send the cancel code and actually ever complete the state transition.

It seems to work just fine as it is?

```
Send: N303 G1 X101.032 Y104.843 E35.58793*95
Changing monitoring state from "Printing" to "Cancelling"
Send: N304 M108*45
Recv: ok
Recv: ok
Send: !!DEBUG:toggle_klipper_connection
Send: N305 M84*25
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: T:210.00/ 210.00 T0:210.00/ 210.00 T1:21.30/ 0.00 B:50.00/ 50.00 @:64
Recv: wait
Send: N306 M104 T0 S0*36
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: wait
Send: N307 M104 T1 S0*36
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: T:210.00/ 210.00 T0:210.00/ 210.00 T1:21.30/ 0.00 B:50.00/ 50.00 @:64
Recv: wait
Send: N308 M140 S0*110
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: wait
Send: N309 M106 S0*109
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: T:210.00/ 210.00 T0:210.00/ 210.00 T1:21.30/ 0.00 B:50.00/ 50.00 @:64
Recv: wait
Changing monitoring state from "Cancelling" to "Operational"
```

foosel, 2021-11-24:
You don't have it sat to wait for the reply to M114 followed by M400, could that be? As in, not recording position on cancel? Because that is part of the usual default behaviour as well and causes it to hang.

cp2004, 2021-11-24:
That does cause a 10 second hang when I enable it - log position on cancel is not enabled by default by the looks of it. Enabling it and printing a file that contained:

```gcode
G28
!!DEBUG: toggle_klipper_connection
G1 X0 Y0
```

Resulted in this output:
```
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 G28*18
Recv: ok
Send: !!DEBUG: toggle_klipper_connection
Send: N2 G1 X0 Y0*43
Recv: !! Lost communication with MCU 'mcu'
Changing monitoring state from "Printing" to "Cancelling"
Send: N3 M108*41
Recv: !! Lost communication with MCU 'mcu'
Send: N4 M400*35
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: wait
Send: N5 M114*34
Recv: !! Lost communication with MCU 'mcu'
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Did not receive parseable position data from printer within 10.0s, continuing without it
Send: N6 M84*25
Recv: !! Lost communication with MCU 'mcu'
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Send: N7 M104 T0 S0*38
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: wait
Send: N8 M104 T1 S0*40
Recv: !! Lost communication with MCU 'mcu'
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Send: N9 M140 S0*108
Recv: !! Lost communication with MCU 'mcu'
Recv: Not SD printing
Recv: wait
Send: N10 M106 S0*86
Recv: !! Lost communication with MCU 'mcu'
Recv: T:21.30/ 0.00 T0:21.30/ 0.00 T1:21.30/ 0.00 B:21.30/ 0.00 @:64
Recv: Not SD printing
Recv: wait
Changing monitoring state from "Cancelling" to "Operational"
```

foosel, 2021-11-24:
Then I have no idea what's different between your test and mine yesterday, it definitely hung for over half an hour before I killed it, I figured it was due to it constantly getting each and every command only errored out and figured that was the reason for it (because position logging etc). Can't test right now because I'm in the middle of debugging another ticket.

Still, even with the cancel running through fully, I'm not too happy with simply leaving things be in such a case. It's in an error state, but the connection doesn't reflect that in any proper way and leaves it to the user to figure out that running a print while the mcu isn't connected won't work. However, if it doesn't actually end up looping forever and my test yesterday was something else, we can at least ignore this for now and figure something out later due to less urgency then.

cp2004, 2021-11-24:
That solution would be good enough for me - Klipper's been around for years, no one has yet had a problem with the connection state that I'm aware of, which is a sign it is not important to change (to me, at least).

For reference, this is what I was testing it with in the config:
```
serial:
  autoconnect: true
  baudrate: 250000
  blacklistedPorts:
  - COM*
  disconnectOnErrors: false
  log: true
  logPositionOnCancel: true
  notifySuppressedCommands: never
```
```
  virtual_printer:
    _config_version: 1
    capabilities:
      Hello: true
    enabled: true
    numExtruders: '2'
```
`OctoPrint 1.8.0.dev147+g4e3258d78`

foosel, 2021-11-30:
Since the consensus seems to be that this works fine for current klipper users (minus the aforementioned and now fixed PrintFailed spamming), we'll leave things otherwise as be and consider this solved with rollout as part of 1.8.0.

cp2004, 2022-05-17:
1.8.0 has been released 🎉

closed 2022-05-17 by cp2004