OctoPrint/OctoPrint #4300: [BUG] Second file uploaded is IGNORED if first is OVERWRITTEN (and second needs to)

bolsoncerrado, 2021-11-04:
### The problem

I try to upload two (or more) files which already exist on the upload folder of octoprint.
For the first uploaded file, I'm asked if I want it overwritten, I choose Yes, overwrite.
The rest of the files I chose to upload are just ignored (not uploaded).

I was expecting the rest of the files I chose to be uploaded (and overwrite or asked) too.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2 (and lowers too, was already there in 1.6.x)

### Operating system running OctoPrint

OctoPi 0.16

### Printer model & used firmware incl. version

Prusa MK3S

### Browser and version of browser, operating system running browser

Chrome 95 on MacOS BigSur

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211104080151.zip](https://github.com/OctoPrint/OctoPrint/files/7472868/octoprint-systeminfo-20211104080151.zip)

jneilliii, 2021-11-04:
Also reported in #4224, but never received additional information. Thank you for your complete bug report.

foosel, 2021-11-04:
So, multiple file uploads in one go are currently not supported. So either this is a bug report about the UI allowing this, or/and a feature request to support this.

cp2004, 2021-11-04:
Uploading multiple files via drag and drop is allowed and works just fine, but only if the files don't collide. If they collide, only the first file is uploaded, which is what this bug report is about.

I just tested this in 1.5.3 to see what the behaviour was before the collision detection stuff, and it was the same, but on collisions they were overwritten on all uploads, as collisions weren't handled previously.

Pressing the 'upload' button only allow you to select a single file, as expected.

foosel, 2021-11-04:
Aaah, ok, yeah the drag and drop is handled by the file upload widget and I guess that will send off several upload requests. Let me rephrase then, officially multiple uploads aren't supported, and that it currently works by drag and drop should be considered a side effect.

bolsoncerrado, 2021-11-04:
So the "Overwrite?" check could be disabled in the settings in the meantime right?

bolsoncerrado, 2022-06-17:
I have the feeling this works as I expected in this latest 1.8.x version? :P

foosel, 2022-06-27:
The overwrite check can now be disabled, however multiple uploads in one go are still not officially supported and that they work is still to be considered a side effect and anything that behaves weirdly for any files but the first one uploaded is a consequence of abusing a possible side effect of the used upload widget, instead of using intentionally supported functionality.

To make sure we are all on the same page here, I'm turning this into a bug report about the drag-n-drop allowing multiple files to be uploaded, which it shouldn't.

cp2004, 2022-06-27:
What would be the cost of making multiple file uploads work correctly? I can see a use for it, and with the file overwrite warning disabled it appears to work perfectly already.

bolsoncerrado, 2022-06-27:
My thoughts exactly. So far 0 issues with this unwanted feature enabled.

Also @foosel been running our farm 24/7 for the last 4 years or so by printing from a unique, single "uploads" folder common to all our instances of Octoprint (spread thru multiple PIs) hassle free as well.  I think this was an unwanted feature as well?

jneilliii, 2022-06-27:
Maybe offload this request to [OctoPrint-MultipleUpload](https://plugins.octoprint.org/plugins/MultipleUpload/), which enables multi file upload with the button as well.

cp2004, 2022-06-27:
> All the problems with GUI support for multi file upload originally raised in https://github.com/OctoPrint/OctoPrint/pull/1868 still exist and would have to be tackled for this to be officially supported.

> However, I'm happy to receive a feature request for this, but that's also something I'd like to see a working PR for that solves all the issues the one linked above did not.

That PR makes much clearer the stuff that would need to be done to make this happen, I hadn't seen it before.

I may consider looking into implementing it, because I think it would be possible to cover the points mentioned. I'll add it to my list to look at the feasibility.

cp2004, 2022-08-06:
This seems feasible, so I'm working on it

cp2004, 2022-09-29:
Multiple file uploads will be supported properly in 1.9.0, since the PR has been merged.

bolsoncerrado, 2022-09-29:
Awesome.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel