OctoPrint/OctoPrint #4301: Double Layer Count on Octoprint

Obicom, 2021-11-04:
### The problem

I use Prusa Slicer and transfer my G-Code via network to Octoprint. Prusa Slicer shows 50 layers for the print (and is correct) Octoprint shows the double of layer (100 layers) since version 1.7.0. I have asked a colleague to double check on his machine and is the same issue.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Makes no sense

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPrint 1.7.2 / Python 3.7.3 / Linux

### Printer model & used firmware incl. version

Prusa MK3S+

### Browser and version of browser, operating system running browser

Windows 10 / Opera Browser

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-11-04:
OctoPrint doesn't count the layers, that's a plugin called DisplayLayerProgress.

It is an issue with the plugin: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/240

Obicom, 2021-11-04:
Please close the issue if there if a ticket with the layer count issue with the plugin is known.

closed 2021-11-04 by Obicom

foosel, 2021-11-04:
Ignore the bot, apparently I overlooked a corner case here.