OctoPrint/OctoPrint #4302: Unable to update to OctoPrint 1.7.2: 'The update did not finish successfully.'

MichMich, 2021-11-04:
### The problem

I'm trying to update OctoPrint from 1.6.1 to 1.7.2 using the Software Update feature.

After pressing the [Update] button using the Stable version. The installation fails with the following message:
```
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

The `plugin_softwareupdate_console.log` reports no issues. The relevant content of `octoprint.log` is af follows:

```
2021-11-04 11:08:42,047 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.7.2...
2021-11-04 11:08:50,630 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1828, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 144, in perform_update
    "Error while executing pip install", (stdout, stderr)
UpdateError: Error while executing pip install
2021-11-04 11:08:50,808 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': '1.7.2', 'from_version': '1.6.1', 'target': u'octoprint'}
```

Thanks for any help regarding this issue!

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Lulzbot Mini

### Browser and version of browser, operating system running browser

Safari and Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211104111058.zip](https://github.com/OctoPrint/OctoPrint/files/7474336/octoprint-systeminfo-20211104111058.zip)

foosel, 2021-11-04:
```
ImportError: cannot import name NullHandler
```

Common issue thanks to a third party plugin pulling in a wrong dependency in a version a long time ago in a galaxy far away, solution pinned to this repo ;)

https://community.octoprint.org/t/my-octoprint-update-fails-with-importerror-cannot-import-name-nullhandler/34493

closed 2021-11-04 by foosel

MichMich, 2021-11-04:
Apologies! Missed that. Works now. Thanks!