OctoPrint/OctoPrint #4305: Since update 1.7.2 Upload via Cura not stored on external folder

SaKiEQ, 2021-11-05:
### The problem

When sending a file from Cura, the file appears in Octoprint Dashboard and can be printed, but on second attempt to print or when returning / restarting Octo the file disappears, while all previous files (before 1.7.2) are still correctly listed. 

I am using external load folder path to a NAS on the same LAN, mounted on the PI. I can see a file being stored on the NAS but with its file name in 'filename.gcode' instead of all previous files filename.gcode. 

As you can see the log contains some error in the last lines, related to Dashboard. 
[octoprint-12.log](https://github.com/OctoPrint/OctoPrint/files/7479995/octoprint-12.log)
[octoprint-systeminfo-20211105045402.zip](https://github.com/OctoPrint/OctoPrint/files/7479996/octoprint-systeminfo-20211105045402.zip)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Cura reports {error: couldn't upload file, invalid type}, which mostly likes is due to a plugin for thumbnails not loaded in safe mode ?

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

Ender 3, stock firmware

### Browser and version of browser, operating system running browser

OS X Safari 15.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

SaKiEQ, 2021-11-05:
I just realized the parentheses in linux just indicate that there spaces in the file name since using file_name.gcode voids those marks. Yet the error both in log and physical interaction with Octoprint persists even when changing the file name in Cura and sending the same file again....

SaKiEQ, 2021-11-05:
I disabled the Cura UFP writer plugin in Cura, so no thumbnails would be be included, but same error in log persists.

jneilliii, 2021-11-05:
The OctoPrint Connection plugin in Cura will transmit the file in ufp format if it detects the Cura Thumbnails plugin installed. I think even if it is disabled. Try exporting your file as gcode and manually uploading, or switch to the Slicer Thumbnails plugin and use the post processing script available in Cura and see if it does the same thing.

jneilliii, 2021-11-05:
But part of the error seems to be permissions related...still plugin error, not OctoPrint. 

> PermissionError: [Errno 13] Permission denied: '/home/pi/NASstorage/Print_Files/Ender3/tmpcahdpeui' -> '/home/pi/NASstorage/Print_Files/Ender3/CE3_Head Gear Clip.gcode'

closed 2021-11-05 by jneilliii

SaKiEQ, 2021-11-05:
Hi, even when uploading a .gcode file directly on the octoprint UI, the same error is thrown in log. See attached loathe last entries. I see that ir refers to the dashboard plugin. 
I read that it refers to a permissions error. Nothing has changed in the 2h from before the update to 1.7.2 after in regards to NAS permissions or user rights on the Pi where OP is running. 
The files are generated on and stored on the NAS, both visible via Linux ls or on the NAS file system it self. Both old files and new have the same pi:root permission/ownership. 

I disabled the Cura plugin in OP too and restarted, the attached log is without Cura plugin loaded and manual gcode upload. 

Not sure what to do more... 

[octoprint-21.log](https://github.com/OctoPrint/OctoPrint/files/7480118/octoprint-21.log)

SaKiEQ, 2021-11-05:
when disabling Dashboard plugin the problem goes away.... so ->Dashboard plugin is the culprit...