OctoPrint/OctoPrint #4307: wrong version reported

sniperlucian, 2021-11-06:
### The problem

expected installed version 1.5.2 
web interface shows update available to 1.7.2
reported version in "software update" panel: 2.0.0.dev831+gd7db47099

 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

2.0.0.dev831+gd7db47099

### Operating system running OctoPrint

manjaro-arm / python 3.9

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.69 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: false
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1296
env.hardware.ram: 1022365696
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.python.pip: 21.1.2
env.python.version: 3.9.5
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 2.0.0.dev831+gd7db47099
printer.firmware: Marlin 2.0.5 (GitHub)

sniperlucian, 2021-11-06:
here was no download link in octoprint to download the bundle. only a copy button - thats why I pasted the system info in the text.

cp2004, 2021-11-06:
How did you install it? Presumably not with `pip install octoprint`? More info is definitely required.

The systeminfo bundle is available starting with 1.6.x, and we don't accept bug reports for older versions, which is why it is required.

foosel, 2021-11-10:
Seconding the question by @cp2004. The only way this could happen is by having installed manually and from a branch that isn't currently mapping to 1.7.x (`master`, `staging/bugfix`) or 1.8.0.dev (`maintenance`, `improve/*`, `fix/*`). Unless you tell us what you did here in as much detail as possible, we can't help you and will have to close this ticket.

sniperlucian, 2021-11-10:
- it had been installed via a standard 1.5 release. I dont recall the source though (pipy / github / pip). 
- octoprint popup in web interface showed that new version is available (1.7.2)
- updating didn't work and further checking revealed that octoprint reported itself beeing on 2.0.0 and update fails (why it reports that a older version (1.7.2 vs. 2.0.0) is an update?)

the file _version.py reports:
```

# This file was generated by 'versioneer.py' (0.15+dev) from
# revision-control system data, or from the parent directory name of an
# unpacked source archive. Distribution tarballs contain a pre-generated copy
# of this file.

import json
import sys

version_json = '''
{
 "dirty": false,
 "error": null,
 "full-revisionid": "gd7db47099",      <---- might be wrong
 "version": "2.0.0"
}
'''  # END VERSION_JSON

def get_versions():
    return json.loads(version_json)

```

foosel, 2021-11-10:
If it's indeed this commit, it's 1.5.3. I have no idea how it ended up reporting as 2.0.0, it certainly wasn't through pypi, it must have been some source checkout done at some point, but since you can't tell us anymore how you got it in the first place it's all guesswork now. I suggest you install a current clean build straight over your install, that should hopefully fix things.

`pip install OctoPrint==1.7.2` in your virtual env should usually do the trick. I sadly cannot tell you where your virtual env is located since I have no idea of your setup.

>     * updating didn't work

How didn't it work? A stack trace goes a long way.

>     * why it reports that a older version (1.7.2 vs. 2.0.0) is an update?

Because you have told it to track stable releases, and thus tries to move you to the latest stable release, not whatever weird state you have currently installed that reports as a wrong version altogether that hasn't even been officially released yet.

Since this doesn't look like it is an OctoPrint issue but rather a weird install, I'm going to close this now since it isn't an active TODO, but commenting is still possible.

closed 2021-11-10 by foosel