OctoPrint/OctoPrint #4309: [Octoprint 1.7.2] Strange behaviour with Octoprint 1.7.2 and using Sub Folders

zebons, 2021-11-07:
I recently upgrade from Octoprint 1.7.0 to 1.7.2 and I see now very strange bahaviour when I launch prints from Sub Folders. 
It wann't the case previously.
I donn't use the SD card to store my Gcode Files and I send it with the drag and drop option directly from my PC thru a browser.
I use Sub Folders to store my Gcode Files.

Since the 1.7.2 upgrade I notice some issues : 
 - When I launch files from Sub Folders, the uploaded time of the file change when the file finish to be printed (of course I donn't re-upload the file...). Sometimes the file status itself isn't good (files correctly printed have normaly green colored label - or red when errors -  but  remain in black or white as they aren't printed with the new version)
 - I use the dashboard plugin and Total layers are false (no value or -1) when the gcode file is stored in a sub folder. If I move that file to the main folder and launch it, total layers value is fine
 - I use the enclosure plugin and light on leds when the print starts and light off when it finish. It doesn't work when I launch  a file stored in a sub folder but no problem when the same file is moved in the main folder

As I said before, all those functions  are working well with the previous version. Now I don't use sub folders to have a correct behaviour but I loved the sub folders option to organize my print files....

zebons, 2021-11-07:
System Info Bundle : 
[
[octoprint-systeminfo-20211107093331.zip](https://github.com/OctoPrint/OctoPrint/files/7491959/octoprint-systeminfo-20211107093331.zip)
](url)

SaKiEQ, 2021-11-07:
Try to update you Dashboard plugin to 1.19.3, came out few hours ago. There is an error in your log pointing to the dashboard plugin. 

We just had a lengthy session of debugging a similar error on my system, which threw the same error, but with different symptoms. 

The above update should fix it.

cp2004, 2021-11-07:
The bug reporting form was not filled out, which means we are missing the **'test in safe mode'** option (as well as the rest of the form).

> When I launch files from Sub Folders, the uploaded time of the file change when the file finish to be printed (of course I donn't re-upload the file...). Sometimes the file status itself isn't good [...]

This does not depend on a plugin like your other two points, so you need to test it in safe mode to verify the same behaviour still occurs.

zebons, 2021-11-07:
Effectively I do a test in safe mode and i donn't notice the same issues. 
I launch a file from a sub folder in safe mode and when the print has finish, the status for the file is good (test colored in green and uplooaded time not changed...)
I'll wait the proposal for upgrading the Dashboard plugin to 1.19.3 and see if it correct my issues.

SaKiEQ, 2021-11-07:
@zebons you can just go to settings / Software Update and use the check for update function....

cp2004, 2021-11-07:
In that case, I am 95% confident that this is an issue with the dashboard plugin, looking at what they've changed recently.

Thank you @SaKiEQ for suggesting that it was the problem.

closed 2021-11-07 by cp2004

SaKiEQ, 2021-11-07:
@cp2004 I spent a bunch of hours over the last 2 days pouring over log files, it definitely looks from the log side of view as the Dashboard (that is not strictly a dashboard problem) bug. At least the 1.19.3 update circumvents that issue.

zebons, 2021-11-07:
@SaKiEQ  I have to wait the proposal for the new Dashboard version

zebons, 2021-11-08:
I do a test after the Dashboard plugin upgrade to 1.19.3 version and I can confirm that all issues are resolved by that new version (green status for files printed, uploaded time, Total layers and even light on/off activated by anothe plugin -Enclosure -).
So that case can be close.
Thanks a lot