OctoPrint/OctoPrint #4310: Upload

rfchsantana, 2021-11-08:
### The problem

Hi, after upgrade 1.7.2 I can't upload files, always give error
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

Octopi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

BloodStasis-cpu, 2021-11-08:
I also have been having this problem
Attached is my systeminfo bundle if it helps
[octoprint-systeminfo-20211108115416.zip](https://github.com/OctoPrint/OctoPrint/files/7498681/octoprint-systeminfo-20211108115416.zip)

cp2004, 2021-11-08:
@BloodStasis-cpu looks like you are running OctoPrint 1.6.1, the current version (that we would be accepting bug reports for) is 1.7.x. You need to update before we can look into bugs.

It also seems that your issue is related to the dashboard plugin, this is where the error in your log is. You should test in safe mode and if the issue does not persist then it is the dashboard plugin.

Can't tell if @rfchsantana has the same issue since they didn't upload the system info bundle.

rfchsantana, 2021-11-08:
[octoprint-systeminfo-20211108180434.zip](https://github.com/OctoPrint/OctoPrint/files/7499146/octoprint-systeminfo-20211108180434.zip)
here is

rfchsantana, 2021-11-08:
{"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}

foosel, 2021-11-08:
That's also caused by the dashboard plugin.

Y'all really should verify an issue persists in safe mode before opening tickets.

rfchsantana, 2021-11-08:
I don't think the problem is dashboard plugin, I also do a fresh install and out the backup, with all plugins as before and worked fine...

closed 2021-11-08 by rfchsantana

foosel, 2021-11-08:
The stack trace for your upload attempt shows the error to be caused by a line inside the dashboard plugin. So yes, it is that. You probably updated it to a fixed version during your reinstall.