OctoPrint/OctoPrint #4312: [Request] Include Klipper's printer.cfg in backups

TonyGiannini, 2021-11-09:
### Describe the solution you'd like

Include Klipper's configuration file ~/printer.cfg in the zipped backup file.  Even better, include any .cfg and .conf files in ~/, i.e. avrdude_gpio.conf.

### Describe alternatives you've considered

It's not difficult to back up and restore the files so this is more of a convenience than anything.

### Additional context

Please close this request if it's already been requested.

jneilliii, 2021-11-09:
similar to my request #4195 which would allow plugins to do this, which would then allow for the OctoKlipper plugin to include it.

foosel, 2021-11-10:
Let's tackle this overall as part of #4195.

closed 2021-11-10 by foosel