OctoPrint/OctoPrint #4315: [Request] Allow selection of which upgrades to perform

KenLucke, 2021-11-18:
### Is your feature request related to a problem? Please describe.

Often, I am running on an older version of a plugin because of an issue, or trying to bug/beta test and have reverted to a previous version.  When OP starts, it always lists upgrades available, which includes the one I manually have regressed to, along with any others.  If I select "Update All," it does just that, including the one I don't want to upgrade, which requires a manual downgrade again.

If you "Ignore" the updates, then go into the "Software Update" screen, you can only do so one by one, with a restart each time, and having to again "Ignore" then "Software Update" again each time.  If there are several updates (which is a not-unusual occurrence, especially if that iteration of OP hasn't been used for a while (I have several printers, and a couple of them are set up specialist for odd-ball filaments or nozzle sizes, so can sometimes sit for a couple of weeks or more between uses).

### Describe the solution you'd like

Two possibilities occur to me:

1) when the list of updates shows, allow a choice of either "Update All" or "Update Selected" - with either checkboxes to select which ones, or being taken to an additional dialog where you can check which ones to update.  Only those marked as desired would be updated.

2) Software Update would allow you to update multiple individual plugins one at a time, and wait until you are finished updating all your choices before restarting (A "Save and Restart Now With Updates" button)

foosel, 2021-11-22:
Have you tried if the existing ignoring function for individual update components/plugins works for your requested workflow?

Clicking the bell in front of each entry allows you to set it to ignore (bell will be struck through), which while ignoring it for notifications will also exclude any available updates from the "Update now" popup.

KenLucke, 2021-11-22:
> Have you tried if the existing ignoring function for individual update components/plugins works for your requested workflow?
>
> Clicking the bell in front of each entry allows you to set it to ignore [...]

I was not even aware of that functionality, as I had never clicked on the bell, nor noticed any documentation on that feature.  Yes, that works just fine.  I will close this request, as that solves my issue.  Thank you.

closed 2021-11-22 by KenLucke