OctoPrint/OctoPrint #4316: Unable to save timelapse temporary directory to anything other than default

Oninaig, 2021-11-20:
### The problem

You are currently unable to save the timelapse temp folder to anything other than the default. If you try, you get an exception in the browser console with a response of: `{"error":"At least one of the configured folders is invalid"}`

This is happening because of line 472 in settings.py: https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/server/api/settings.py#L472

You are trying to retrieve the key `timelapse_tmp` from the data dictionary but that key doesn't exist because you are not using the mapping dictionary defined on line 26 but instead just using the `FOLDER_TYPES` key `timelapse_tmp`. Meanwhile the actual key in `data` is `timelapseTmp`.

My logs will show the error on a different line than src because I added a bunch of debug lines.

[octoprint-systeminfo-20211120095509.zip](https://github.com/OctoPrint/OctoPrint/files/7574710/octoprint-systeminfo-20211120095509.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

Octopi 0.18.0

### Printer model & used firmware incl. version

Prusa mk3s+

### Browser and version of browser, operating system running browser

Chrome latest 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

JS response:

```
packed_libs.js?1fe9f29c:13 POST http://192.168.1.101/api/settings 400 (BAD REQUEST)
send @ packed_libs.js?1fe9f29c:13
ajax @ packed_libs.js?1fe9f29c:13
OctoPrintClient.ajax @ packed_client.js?b2d27cf4:8
OctoPrintClient.ajaxWithData @ packed_client.js?b2d27cf4:8
OctoPrintClient.post @ packed_client.js?b2d27cf4:8
OctoPrintClient.postJson @ packed_client.js?b2d27cf4:8
OctoPrintSettingsClient.save @ packed_client.js?b2d27cf4:103
SettingsViewModel.self.saveData @ packed_core.js?0d093a46:336
eval @ VM9148:3
(anonymous) @ packed_libs.js?1fe9f29c:133
dispatch @ packed_libs.js?1fe9f29c:13
v.handle @ packed_libs.js?1fe9f29c:13
```

cp2004, 2021-11-20:
Confirmed this one in discussion on Discord the other day. Haven't tested when the last it did work was.

foosel, 2021-11-23:
Fixed, stupid logic error. Ready for 1.8.0.

The last time this worked must have been 1.4.2, because the offending commit b96f7657e68333545ac7d05eef364af8cce8528b got introduced in 1.5.0rc1 and the lines in question hadn't been touched since. So apparently no one tried modifying the timelapse tmp folder for a year, or if so didn't report it. Thanks @Oninaig for giving us the heads-up so this could get solved :)

commit 4b9a1c2c (2021-11-23): 🐛 Fix timelapse tmp folder configuration

Closes #4316
closed 2022-05-17 by foosel