OctoPrint/OctoPrint #4317: Klipper command BED_MESH_CALIBRATE in Long Running Commands ignored.

lucky62, 2021-11-20:
### The problem

I am using the OctoPrint (OctoPi on Rpi3) with Klipper.
I have configured Long running commands according this picture.

[image]

But OctoPrint is still reporting the Timeouts when BED_MESH_CALIBRATE is executed.

Is this important? **Just the G or M code, comma separated.**

Should I understand this, that I cannot put this command here, because it is not G nor M command?
[octoprint-systeminfo-20211120174246.zip](https://github.com/OctoPrint/OctoPrint/files/7574956/octoprint-systeminfo-20211120174246.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

DIY printer with BTT SKR V1.4

### Browser and version of browser, operating system running browser

Firefox, Linux Mint 20.2 Uma

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-11-20:
> Is this important? Just the G or M code, comma separated.

> Should I understand this, that I cannot put this command here, because it is not G nor M command?

That is indeed the problem here, `BED_MESH_CALIBRATE` is not a gcode command and so it's not recognised by the communication layer to compare it to this list.

Another solution is probably needed to make sure there's no timeouts when using macros. This is probably an issue with Marlin's new macros as well, but they look like gcode (M810-M819) so can be used in the list.

jneilliii, 2021-11-20:
Can't you also configure G29 as macro in klipper to run the command as an alternative?

foosel, 2021-11-23:
As already said, long running commands currently target actual GCODE, not Klipper macros.

In order to figure out what the correct course of action here is, I'd like to know whether Klipper macros generate `ok` messages on completion, tolerate line numbering and check summing and in general behave identical to GCODE? Also, I'd like you, @lucky62, to enable the serial.log, reproduce the issue and then share the resulting systeminfo bundle, because that might also shed some light on these questions.

lucky62, 2021-11-25:
Hi @foosel, 
BED_MESH_CALIBRATE is not the klipper macro. It is a normal klipper command.
Just it is not starting with G or M... Klipper has many [extended-g-code-commands](https://www.klipper3d.org/G-Codes.html#extended-g-code-commands)...

I think the [klipper macros](https://www.klipper3d.org/Config_Reference.html#g-code-macros-and-events) (defined in the firmware config file - e.g. printer.cfg - not in the OctoKlipper plugin) are as normal G-code commands (they are returning ok, when finished).
Most probably this macro can solve my problem:
```
[gcode_macro G29]
BED_MESH_CALIBRATE
```
Unfortunately now I cannot help much. I re-flashed my printer with Marling. I am testing Marlin to see what is the better for me...

foosel, 2021-11-30:
Non-GCODE commands should now also be supported in long running commands (as long as they *don't* have parameters or anything like that, as I can only parse that off of GCODE lines). Will be part of 1.8.0.

In any case, firmware in general should implement the [busy protocol](https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer) for long running commands in the background to keep a working heartbeat for OctoPrint to detect that the printer is in fact still alive, instead of just going AWOL for undetermined amounts of time on lengthy macros or anything like that.

commit b32f23e7 (2021-11-30): ✨ Expand long running commands to support non-GCODE

Closes #4317
closed 2022-05-17 by foosel