OctoPrint/OctoPrint #4326: [Request] Ability to ignore Printer Notifications based on regex

dburr, 2021-11-29:
### Is your feature request related to a problem? Please describe.

I use the [Detailed Progress](https://plugins.octoprint.org/plugins/detailedprogress/) plugin to update my printer's display with the estimated time to completion, etc. It does this by periodically injecting `M117` commands to the printer. Unfortunately since `M117` commands are echoed back as notification actions, this results in the "Printer Notifications" list getting filled with useless/trivial notifications such as this.
[image]

### Describe the solution you'd like

It would be nice to be able to conditionally ignore notifications from the printer based on a regex. So, for example, to filter out the useless messages in my screenshot above, I would use regexes like:

`^\d+%`
`"Layer \d+"`
`"Done In .*"`
`"^(Mon|Tue|Wed|Thu|Fri|Sat|Sun).*"`

### Describe alternatives you've considered

Obviously I could disable the Detailed Progress plugin, but then I would miss out on that useful information being shown on the printer (which I do check periodically)

foosel, 2021-11-29:
> Unfortunately since `M117` commands are echoed back as notification actions

TBH, that sounds more like the firmware is abusing the printer notification feature. It's meant for notifications that are NOT triggered by the host itself but would otherwise require the user to stand in front of the printer to even have a knowledge of. Sadly this feature has seen a TON of abuse like this ever since I added it to OctoPrint, to a point where I wonder whether it was a good idea to add it at all, with firmware happily spamming its users through it.

We can look into a filter here, but **please** also report this to your firmware issue tracker so that may look into fixing this constant spam on their end as well.

jneilliii, 2021-11-29:
Marlin has been doing this for a while now. I think it may have been utilizing this to support the newer touch screen interfaces.

foosel, 2021-11-29:
Doesn't change that it's abusing the feature. This isn't the first time I've had to reduce its usability and simplicity on my end due to it getting abused and as a consequence spamming the user (the only reason why there even is a notification AREA now instead of only popups was another case of heavy abuse). It is starting to get ridiculous.

lscarneiro, 2022-04-14:
Is there a way we can at least limit the messages? Like, "keep last `x`"? 

Filtering is definitely a plus, but if we're able to at least limit the number of messages kept, this would help on long prints as the UI gets slow with too many elements on the page.

foosel, 2022-04-14:
I can look into this but **please** also address this with Marlin. This feature is getting abused. I'm forced to reduce its usefulness again and again due to the constant message spam that gets more and more from firmware version to version. I'm currently frankly leaning towards unbundling the related plugin as a consequence. This needs to be addressed on the firmware side as well.

lscarneiro, 2022-04-14:
Isn't it a problem with how this is fired back from `Detailed Progress` / `DisplayLayerProgress` plugins?

I use both plugins (`Detailed Progress` actually uses `DisplayLayerProgress` as a requirement) which looks like is the case for the OP as well.

I can be totally wrong, but I feel like it should be possible for this communication to be filtered out internally (with some new implementation probably) since this relies on `M117` and it is fired from a plugin, maybe there's a way we can work between the plugins and OctoPrint to filter this out. 
I'm just not seeing how Marlin can intervene if this is an `M117` situation, and the bundled plugin spits all the messages to the UI, so how can Marlin know it shouldn't report back to OctoPrint (if this is even the case, sorry, I don't know how this interaction is implemented)

jneilliii, 2022-04-14:
The issue isn't so much the usage of M117 on the firmware side, but the use of action notifications in response to M117 commands. M117 is used to display a message on the LCD screen and is not intended to be an action notification because it's not an action. Marlin was doing this IIRC in order to support the touch screen interfaces and have them display messages on those screens when M117 messages are sent.

lscarneiro, 2022-04-14:
I see... thanks for the clarification 😊 

So, probably the "simplest solution" (not saying it's easy, though), would be to be able to configure a limit of messages shown, then?

cp2004, 2022-04-14:
There's probably a case for a limit such as 30-50 messages? To stop it from getting overwhelmed and causing a performance issue. I just guessed that number out of the air.

The `action:notification` system is not just an OctoPrint specific feature, as @jneilliii says it's useful for touchscreens acting as a host. The issue of 'spam' here is in part caused by OctoPrint plugins sending lots of M117 updates - usually a screen would just show the latest one, here we save them. I'm not going to argue whether that was the original intention of the feature or not.

There was a PR merged recently to allow some configuration of how many notifications are sent (https://github.com/MarlinFirmware/Marlin/pull/22833). Some of the pushback mentioned was further discussed on discord at the time.

foosel, 2022-05-11:
The problem with a limit that I see is that all the spam could then simply flood out the actually meaningful messages. From that perspective a filter regex or something would indeed make more sense.

---

Implementation idea:

- [ ] new filter setting in [`src/octoprint/plugins/action_command_notification/__init__.py`](https://github.com/OctoPrint/OctoPrint/blob/500729607d00069e65ca2105f5af42919d861792/src/octoprint/plugins/action_command_notification/__init__.py#L62-L63)
- [ ] apply filter against incoming messages: [`src/octoprint/plugins/action_command_notification/__init__.py`](https://github.com/OctoPrint/OctoPrint/blob/500729607d00069e65ca2105f5af42919d861792/src/octoprint/plugins/action_command_notification/__init__.py#L108-L126)
- [ ] expose filter setting on settings UI: [`src/octoprint/plugins/action_command_notification/templates/action_command_notification_settings.jinja2`](https://github.com/OctoPrint/OctoPrint/blob/500729607d00069e65ca2105f5af42919d861792/src/octoprint/plugins/action_command_notification/templates/action_command_notification_settings.jinja2)

jatin-47, 2023-09-16:
I want to work on this issue. I see that @DShenkle had opened a PR but he is no longer working on that PR so, should I start working afresh or continue where he left off?

jneilliii, 2023-09-16:
I have a working plugin that does the ignoring part, just not based on regex, but exact match...

https://github.com/jneilliii/OctoPrint-FirmwareErrorHandler

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel