OctoPrint/OctoPrint #4328: [Request] IDEX Support for Duplicate and Mirror Mode

Parker-Bergen, 2021-12-03:
### Describe the solution you'd like

I am using a Flash Forge Creator 2, which is an IDEX printer that supports mirror mode and duplicate modes when printing. Octoprint will allow both nozzles to heat up but only moves and extrudes one print head. It would be nice to have support for both to move and print since the gcode being output from FlashPrint 5 which contains the proper calls for this.

Parker-Bergen, 2021-12-03:
As I'm looking at my Gcode it seems that it does not contain the correct calls for duplicate or mirror mode. Does octoprint support duplicate or mirror mode already?

cp2004, 2021-12-03:
OctoPrint (mostly) just sends the gcode it's been given to the printer line by line - if you put the commands in your gcode they will be sent to the printer.

Some plugins might have an issues with the IDEX/duplication stuff if they make some assumptions, but that's obviously an issue for them not OctoPrint core.

Having said that, not sure what the gcode viewer would do. That is just cosmetic though.

Parker-Bergen, 2021-12-09:
Currently solving the issue with the plugin I am using: OctoPrint-FlashForge

closed 2021-12-09 by Parker-Bergen