OctoPrint/OctoPrint #4329: Custom controls output shows next to wrong button

arim215, 2021-12-03:
### The problem

I added 2 costum controls with output in my .octoprint/config.yaml file and both have regex filter.
When pressing the button for the first control with output, the output shows up next to the second button with the regex of the second button. But pressing the second button, all shows up correctl
[image]
y
[image]
.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Ender 3 Pro - Marlin 2.0.x-bugfix

### Browser and version of browser, operating system running browser

Google Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20211202230637.zip](https://github.com/OctoPrint/OctoPrint/files/7646742/octoprint-systeminfo-20211202230637.zip
[Config.yaml and gcode script.zip](https://github.com/OctoPrint/OctoPrint/files/7646743/Config.yaml.and.gcode.script.zip)
)

cp2004, 2021-12-04:
Could you copy the custom controls configuration you are using? So that someone could try and reproduce the issue and figure out what's wrong. Don't share the whole of the config.yaml file as it contains secrets.

arim215, 2021-12-05:
> 

I have put the "controls" portion of the config.yaml file in the original post, but here it is again (thank you for telling me not to put the hole file, as i originally did, but removed it
[config.yaml.txt](https://github.com/OctoPrint/OctoPrint/files/7655138/config.yaml.txt)
)

foosel, 2022-01-17:
Your Z offset regex matches the output of `M92` as well, so OctoPrint doesn't know where it belongs. There are certain firmware builds out there which have a bug that cause commands to be acknowledged in a different order than received, which is why OctoPrint can't match up request/response through order and has to resort to pure pattern matching on received lines. In your case here that matching has an overlap, and your z-offset regex wins.

Try further specifying the regexes. Both response lines come prefixed with the command you sent, so also match against THAT.
E.g. for your E-steps, try (untested!):

```
M92.*?E([-+]?[0-9.]+).*?
```

and for your z-offset

```
M851.*?Z([-+]?[0-9.]+).*?
```

Always match as specific as possible.

Please report back on if this solves the issue.

arim215, 2022-01-18:
I can confirm this fix works! thank you

closed 2022-01-18 by arim215