OctoPrint/OctoPrint #4330: [Request] Log action command pause

cp2004, 2021-12-04:
### Is your feature request related to a problem? Please describe.

If a printer pauses the print without the user asking it to - for example, on the Prusa MK3S it will pause if the fan speeds are too low - it can leave users clueless. Likewise I guess if a filament sensor triggered.

OctoPrint logs the reason for pausing/cancelling from API calls and plugins, but it doesn't seem to from action commands. This would allow for a quick explanation. 'Pausing print on behalf of the printer (action command)' means one doesn't have to spend ages guessing/enabling serial.log and hoping it might happen again.

### Describe the solution you'd like

Log something like the message above, on pause from `// action:pause` from the printer.

### Additional context

https://discord.com/channels/704958479194128507/705046975669600298/916817893839216680

evanwurden, 2022-12-04:
Is this still an issue? I see "Pausing on request of the printer..." logged when the printer is paused via action command.

cp2004, 2022-12-16:
Indeed it is no longer an issue, since it was addressed in #4358. I hadn't realised that when adding the help for a contributor! Thanks for pointing that out.

closed 2022-12-16 by cp2004