OctoPrint/OctoPrint #4331: [Request] Keep steppers enabled on print pauses

your-friend-alice, 2021-12-07:
### Is your feature request related to a problem? Please describe.

I insert pauses (`M601` from prusaslicer, added to the pause print commands setting) into my prints in order to embed objects like nuts or magnets into the print. It's easy to jostle things a bit while working, and the slightest bump knocks things out of alignment and ruins the print. When printing directly from USB without OctoPrint, I don't have this problem. I do not have any `M18` or `M84` commands in my pause GCODE scrips, so disabling the steppers appears to be a hardcoded behavior in Octoprint.

Other people seem to have complained about this on [reddit](https://www.reddit.com/r/octoprint/comments/au9r3g/octoprint_disable_stepper_motor_on_pause_print/).

### Describe the solution you'd like

I'd like a setting to disable this behavior, and leave the steppers enabled during print pauses.

### Describe alternatives you've considered

Sometimes it's possible to recover from a Y axis bump on my Prusa Mini by hitting the button in the Octoprint Control tab to home the XY axes, but this seems error prone and frequently fails.

### Additional context

My settings. Should not be relevant, but should eliminate some possibilities:

GCODE Scripts / After print job is paused:
```
G91 Z ; Set to relative positioning mode
G1 Z20 E-10; Move hotend 20mm up from part
G90 Z ; Set to absolute positioning mode
M83 ; Set extruder to relative
G60 S0 ; Save current position
G1 X180 Y180 ; Move the bed to the front
M17 ; Enable steppers (they seem to get disabled for some reason)
```
GCODE Scripts / Before print job is resumed:
```
G61 XY S0 ; Restore XY positions
G91 Z ; Set to relative positioning mode
G1 Z20 E10; Move hotend 20mm down to part
G90 Z ; Set to absolute positioning mode
M83 ; Set extruder to relative
```
Serial Connection / Firmware & protocol / Pausing commands:
```
M0, M1, M25, M601
```

foosel, 2021-12-07:
OctoPrint doesn't disable the steppers on pause. It only does so on cancel, and only if this is configured in the cancel script. OctoPrint will sent what is in the pause and resume scripts, and in case of SD prints also the relevant SD commands to actually pause and resume, and this is it.

I remember there was an issue with M601 recently making its rounds on the forums where it would disable the heaters after a while. If I were you I'd check what M601 is actually doing on the printer side to rule it out as the issue here.

foosel, 2021-12-07:
I should add that this applies to stock OctoPrint, I can't vouch for what any third party plugins you might have installed might be doing, so I also highly suggest you check the behaviour in safe mode.

cp2004, 2021-12-07:
Enabling the serial.log would mean you could show that there are no M18 commands being sent to the printer on pause.

The issue we've seen with M601 before is that Prusa disabled the heaters when paused, and then they were not automatically re-enabled when printing (re)started. So I would not put it beyond their firmware to also disable the steppers on an M601 command.