OctoPrint/OctoPrint #4332: [Request] Software Update: Improve logging in case of failing update checks

foosel, 2021-12-07:
### Is your feature request related to a problem? Please describe.

As witnessed on the forums [here](https://community.octoprint.org/t/not-being-offered-upgrades/38829) there's currently insufficient logging in case of failures in running update checks, e.g. GitHub release fetching with an outdated PAT. The "available" version simply stays empty without any log message to indicate why.

### Describe the solution you'd like

A failing request should at least cause a log entry, not just a silent error. That would allow easier debugging of a failure to check for updates. Better yet would be to log the reason for the error in the frontend, like with rate limiting situations. Something like "Backend request failed, see log" would already be more helpful than the current sitation.

### Additional context

https://community.octoprint.org/t/not-being-offered-upgrades/38829

foosel, 2022-02-28:
This will be in 1.8.0

cp2004, 2022-05-17:
1.8.0 has been released 🎉

closed 2022-05-17 by cp2004