OctoPrint/OctoPrint #4333: Not queuing T0

DiscusZ, 2021-12-07:
Not queuing T0, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware. Make sure your printer profile is set up correctly.

I have been hit by this error on Octoprint, the only changes that have been done is upgrading Octoprint.  I have no upgraded Marlin firmware in a long time as my printers have been working great.

I looked back at my previous Gcode and  what throws the error is this

; process Process1
; layer 1, Z = 0.240
T0
; tool H0.240 W0.576

This was pulled from a  item I printed back in Aug 2021.

I am using S3D 4.x

Why did this all of a sudden start throwing an error?

I am running FIRMWARE_NAME:Marlin 1.1.9 (Github)

cp2004, 2021-12-07:
According to the log file, at some points when you started a print your printer reported both the `T0` and `T1` commands to be invalid.

```
2021-12-07 03:30:06,154 - octoprint.util.comm - WARNING - T0 reported as invalid, reverting to T0
2021-12-07 03:30:06,180 - octoprint.util.comm - WARNING - T1 reported as invalid, reverting to T0
```

The only way to get a more detailed error message would be to **enable serial logging** and then **upload the serial.log file**.

cp2004, 2021-12-26:
Due to a lack of a response to this issue, it will be closed.

closed 2021-12-26 by cp2004