OctoPrint/OctoPrint #4335: [Request] keep last good values on screen when a fault occurs

yoyo42, 2021-12-08:
### Is your feature request related to a problem? Please describe.

My printer has started occasionally failing with an "M112 Emergency Stop" recently, probably due to a wiring issue. However recovering from this is made a lot harder because I don't know what Z height it was at to resume the print, because Octoprint disconnects and clears all the values from the screen.

The terminal log only goes back 300 lines which isn't enough to see the previous layer change G-code.

### Describe the solution you'd like

Please could the last known good values for position be kept visible and/or logged, maybe as a notification e.g. 'M112 Emergency stop called at X:100.0 Y100.0 Z123.5'

This would make it much easier to resume the print or just reprint the top half from that Z upwards to patch it up.

These values are displayed (and cleared) by the Octoprint core so I believe that's where the change would need to be made.

### Describe alternatives you've considered

I will also be suggesting the fail position is added to the printer LCD instead of just showing 'M112 Emergency Stop' but that is printer specific, doing it in Octoprint makes it more generally useful.

cp2004, 2021-12-15:
Duplicate of #4264 (or similar problem likely covered by it)

closed 2021-12-15 by cp2004

yoyo42, 2021-12-19:
You're right that I'm looking at the printer screen, but Octoprint is sending the M112 Emergency Stop that causes the screen to be wiped. I agree that Octoprint knows the information I'd like to see so dealing with it there is better.