OctoPrint/OctoPrint #4341: [Request] Show uploaded prints in the Files list

thijstriemstra, 2021-12-12:
### Is your feature request related to a problem? Please describe.

When I sent a file from Prusa Slicr (using 2.4.0beta4) to octoprint, the file doesn't show up in the Files list on the octoprint dashboard. Which is a shame, I really like the list of completed/failed prints with thumbnail.

### Describe the solution you'd like

Show prints uploaded from external program like Prusa Slicr in the Files list.

foosel, 2021-12-12:
Uploaded files should always be in said list, but maybe in a sub folder. Check any folders you might have created, and if that doesn't help provide precise reproduction steps, try in safe mode and share a system info bundle.

thijstriemstra, 2021-12-12:
> Uploaded files should always be in said list, but maybe in a sub folder.

Sorry I think I shouldn't have used the word 'uploaded' because Prusa Slicr uses an API that obviously uploads the gcode file but octoprint doesn't list it in the Files section.

This is the Prusa Slicr dialog:

[image]

The Upload or Upload and print buttons don't store them in the Files list (although it's visible in Current Printing in top panel):

[image]

thijstriemstra, 2021-12-12:
The gcode file is present in `.octoprint/uploads/`:

```
pi@octopi:~ $ ls -l .octoprint/uploads/
-rw-r--r-- 1 pi pi  5850820 dec 12 20:55 'camera arm support_0.2mm_PLA_MINI_2h48m.gcode'
-rw-r--r-- 1 pi pi  2202762 dec 12 19:17 'camera insert_0.2mm_PLA_MINI_1h35m.gcode'
-rw-r--r-- 1 pi pi   260515 nov 25 20:39  cap_0.15mm_PETG_MINI_5m.gcode
-rw-r--r-- 1 pi pi 13597460 dec  4 03:19  case-body_0.25mm_PLA_MINI_8h51m.gcode
-rw-r--r-- 1 pi pi 13663786 dec  4 03:21  case-body_0.25mm_PLA_MINI_8h57m.gcode
-rw-r--r-- 1 pi pi  4260063 dec  4 01:54  case-body_0.2mm_ABS_MINI_1h54m.gcode
```

foosel, 2021-12-12:
Try scrolling down the files list please. All I see is a file starting with a, your current starting with c and possible during by name applied indicating it this will not show up at the top of the list.

The API that the slicer uses is the exact same that is used if you upload through the front-end. There is no difference. And OctoPrint can't print local files that aren't in the files list, it's simply impossible. What you are asking for has literally been how OctoPrint has worked since day one.

So if it doesn't for you, you either haven't scrolled down, have accidentally uploaded to a folder, some third party plugin is interfering (safe mode!) or you are running into a weird browser issue (system info bundle!).

jneilliii, 2021-12-12:
make sure you don't have your file list filtered to only show SD files.

thijstriemstra, 2021-12-12:
Thanks for the feedback!

Here are the Files sort settings:

[image]

Scrolling down the list only shows older files.

> There is no difference. And OctoPrint can't print local files that aren't in the files list, it's simply impossible. What you are asking for has literally been how OctoPrint has worked since day one.

Gotcha, and that's why I was hesitant of filing this bug but here we are. Do you use Prusa Slicr, has anyone else encountered this? What version of Prusa Slicr are you using?

> weird browser issue (system info bundle!).

Here it is:

[octoprint-systeminfo-20211212222725.zip](https://github.com/OctoPrint/OctoPrint/files/7699894/octoprint-systeminfo-20211212222725.zip)

thijstriemstra, 2021-12-12:
Oh jezus. I changed sort settings and it's working now. Embarrassing!!

closed 2021-12-12 by thijstriemstra

jneilliii, 2021-12-12:
you probably could have simply pressed the little refresh icon and they would show up.

thijstriemstra, 2021-12-12:
I did while filing this report, but no, it didn't change anything in the list.