OctoPrint/OctoPrint #4342: Updated to 2.7.2 and getting "ReferenceError: UAParser is not defined"

canda, 2021-12-12:
### The problem

I had octoprint version 1.6
Updated into version 1.7.2
When I try to access octoprint via the browser I'm stuck on "Connecting to OctoPrint's server..."
And I see this error on javascript console
[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I cannot access the octoprint ui to click on "restart OctoPrint on safe mode"

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

unknown, I can't access the ui

### Printer model & used firmware incl. version

ender3

### Browser and version of browser, operating system running browser

chrome 96.0.4664.93, macOS 10.15.6

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

packed_libs.js?9b63d90f:13 Uncaught ReferenceError: UAParser is not defined
    at packed_core.js?6d618f2d:1218
    at HTMLDocument.<anonymous> (packed_core.js?6d618f2d:1223)
    at e (packed_libs.js?9b63d90f:13)
    at t (packed_libs.js?9b63d90f:13)
(anonymous) @ packed_core.js?6d618f2d:1218
(anonymous) @ packed_core.js?6d618f2d:1223
e @ packed_libs.js?9b63d90f:13
t @ packed_libs.js?9b63d90f:13
setTimeout (async)
S.readyException @ packed_libs.js?9b63d90f:13
(anonymous) @ packed_libs.js?9b63d90f:13
e @ packed_libs.js?9b63d90f:13
t @ packed_libs.js?9b63d90f:13
setTimeout (async)
(anonymous) @ packed_libs.js?9b63d90f:13
c @ packed_libs.js?9b63d90f:13
fireWith @ packed_libs.js?9b63d90f:13
fire @ packed_libs.js?9b63d90f:13
c @ packed_libs.js?9b63d90f:13
fireWith @ packed_libs.js?9b63d90f:13
t @ packed_libs.js?9b63d90f:13
setTimeout (async)
(anonymous) @ packed_libs.js?9b63d90f:13
c @ packed_libs.js?9b63d90f:13
fireWith @ packed_libs.js?9b63d90f:13
fire @ packed_libs.js?9b63d90f:13
c @ packed_libs.js?9b63d90f:13
fireWith @ packed_libs.js?9b63d90f:13
ready @ packed_libs.js?9b63d90f:13
B @ packed_libs.js?9b63d90f:13

canda, 2021-12-12:
I can't access the system info because the UI is unresponsive

foosel, 2021-12-12:
The link explains how to retrieve a bundle even if the UI is unresponsive. Eg via the recovery page at /recovery, or the command line.

cp2004, 2021-12-12:
Clearing the cache might help.

canda, 2021-12-12:
Ah, got it. `/recovery` worked
Here is my systeminfo bundle
[octoprint-systeminfo-20211212185727.zip](https://github.com/OctoPrint/OctoPrint/files/7699935/octoprint-systeminfo-20211212185727.zip)

> Clearing the cache might help.

Clearing the browser cache did not help.
I even tried from an incognito window. It asked me to log in. After logging in, I get the same error.

cp2004, 2021-12-12:
Looks like the 1.7.2 upgrade didn't go as planned - it's still reporting 1.6.1, and missing some static files.

It looks like it might be installed system wide, rather than in a virtual environment - maybe this messed with the upgrade slightly? It shouldn't do, but the recommended way to install OctoPrint is using a virtual environment.

You could attempt to reinstall the update, from SSH - `pip install --force-reinstall octoprint`.

Also note that you are using Python 2, [you should upgrade it as soon as possible](https://community.octoprint.org/t/upgrading-your-octoprint-install-to-python-3/35158).

foosel, 2021-12-15:
Closing as a corrupted install is not a bug in OctoPrint but a corrupted install.

closed 2021-12-15 by foosel