OctoPrint/OctoPrint #4343: gcodeviewer ignored in config.yaml appearance-->components items

synman, 2021-12-13:
### The problem

Apologies for not catching the version this regressed in.  I'm going to guess it was somewhere between 1.4 and 1.5.  

We've lost the ability to disable and reorder the gcode tab via appearance/components/disabled and order.   This syntax works fine for all other tabs but, as stated above, about a year ago this broke.  Apologies for not reporting this sooner.  I opened an issue on my own plugin when I first detected it back in January 2021 and just now got around to looking into it.

https://github.com/synman/Octoprint-Bettergrblsupport/issues/32

I tried variations on "gcodeviewer" thinking maybe you changed the name to something like "gcode" but that did not make a difference after putzing around with it this evening.

```
appearance:
  color: black
  components:
    disabled:
      tab:
      - temperature
      - control
      - gcodeviewer
    order:
      tab:
      - plugin_bettergrblsupport
      - temperature
      - control
      - gcodeviewer
      - terminal
      - timelapse
```
   

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

validated as broken in safe mode too

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPi 0.16.0

### Printer model & used firmware incl. version

Eleksmaker A3 Laser Engraver (GRBL)

### Browser and version of browser, operating system running browser

Latet Safari, OSX Monteray 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

This is pretty cut and dry.  I'm more than happy to supply additional details but am unsure how any of the items listed here would prove helpful.

synman, 2021-12-13:
you're welcome, silly bot

synman, 2021-12-13:
so I did a dump of all the views as views.py is iterating thru them:

```
2021-12-12 21:51:18,585 - octoprint.server.views - WARNING - content tab: about
2021-12-12 21:51:18,590 - octoprint.server.views - WARNING - content tab: plugin_pi_support
2021-12-12 21:51:18,591 - octoprint.server.views - WARNING - content tab: supporters
2021-12-12 21:51:18,597 - octoprint.server.views - WARNING - content tab: authors
2021-12-12 21:51:18,603 - octoprint.server.views - WARNING - content tab: license
2021-12-12 21:51:18,607 - octoprint.server.views - WARNING - content tab: thirdparty
2021-12-12 21:51:18,615 - octoprint.server.views - WARNING - content tab: plugin_pluginmanager
2021-12-12 21:51:18,637 - octoprint.server.views - WARNING - content tab: access
2021-12-12 21:51:18,639 - octoprint.server.views - WARNING - content tab: interface
2021-12-12 21:51:18,652 - octoprint.server.views - WARNING - content tab: plugin_bettergrblsupport
2021-12-12 21:51:18,654 - octoprint.server.views - WARNING - content tab: terminal
2021-12-12 21:51:18,655 - octoprint.server.views - WARNING - content tab: timelapse
2021-12-12 21:51:18,672 - octoprint.server.views - WARNING - content tab: section_printer
2021-12-12 21:51:18,673 - octoprint.server.views - WARNING - content tab: serial
2021-12-12 21:51:18,675 - octoprint.server.views - WARNING - content tab: printerprofiles
2021-12-12 21:51:18,682 - octoprint.server.views - WARNING - content tab: temperatures
2021-12-12 21:51:18,684 - octoprint.server.views - WARNING - content tab: terminalfilters
2021-12-12 21:51:18,743 - octoprint.server.views - WARNING - content tab: gcodescripts
2021-12-12 21:51:18,805 - octoprint.server.views - WARNING - content tab: section_features
2021-12-12 21:51:18,878 - octoprint.server.views - WARNING - content tab: features
2021-12-12 21:51:18,904 - octoprint.server.views - WARNING - content tab: webcam
2021-12-12 21:51:18,920 - octoprint.server.views - WARNING - content tab: accesscontrol
2021-12-12 21:51:18,946 - octoprint.server.views - WARNING - content tab: plugin_gcodeviewer
2021-12-12 21:51:18,950 - octoprint.server.views - WARNING - content tab: api
2021-12-12 21:51:18,951 - octoprint.server.views - WARNING - content tab: plugin_appkeys
2021-12-12 21:51:18,952 - octoprint.server.views - WARNING - content tab: section_octoprint
2021-12-12 21:51:18,955 - octoprint.server.views - WARNING - content tab: server
2021-12-12 21:51:18,969 - octoprint.server.views - WARNING - content tab: folders
2021-12-12 21:51:18,974 - octoprint.server.views - WARNING - content tab: appearance
2021-12-12 21:51:18,978 - octoprint.server.views - WARNING - content tab: plugin_logging
2021-12-12 21:51:18,979 - octoprint.server.views - WARNING - content tab: plugin_pluginmanager
2021-12-12 21:51:18,983 - octoprint.server.views - WARNING - content tab: plugin_softwareupdate
2021-12-12 21:51:18,987 - octoprint.server.views - WARNING - content tab: plugin_announcements
2021-12-12 21:51:19,003 - octoprint.server.views - WARNING - content tab: plugin_eventmanager
2021-12-12 21:51:19,005 - octoprint.server.views - WARNING - content tab: plugin_backup
2021-12-12 21:51:19,011 - octoprint.server.views - WARNING - content tab: plugin_tracking
2021-12-12 21:51:19,017 - octoprint.server.views - WARNING - content tab: plugin_errortracking
2021-12-12 21:51:19,019 - octoprint.server.views - WARNING - content tab: plugin_pi_support
2021-12-12 21:51:19,033 - octoprint.server.views - WARNING - content tab: settings
2021-12-12 21:51:19,034 - octoprint.server.views - WARNING - content tab: systemmenu
2021-12-12 21:51:19,035 - octoprint.server.views - WARNING - content tab: plugin_announcements
2021-12-12 21:51:19,038 - octoprint.server.views - WARNING - content tab: plugin_logging_seriallog
2021-12-12 21:51:19,040 - octoprint.server.views - WARNING - content tab: plugin_logging_plugintimingslog
2021-12-12 21:51:19,050 - octoprint.server.views - WARNING - content tab: plugin_pi_support
2021-12-12 21:51:19,054 - octoprint.server.views - WARNING - content tab: login
2021-12-12 21:51:19,067 - octoprint.server.views - WARNING - content tab: plugin_firmware_check_warning
2021-12-12 21:51:19,068 - octoprint.server.views - WARNING - content tab: plugin_firmware_check_info
2021-12-12 21:51:19,073 - octoprint.server.views - WARNING - content tab: connection
2021-12-12 21:51:19,075 - octoprint.server.views - WARNING - content tab: state
2021-12-12 21:51:19,097 - octoprint.server.views - WARNING - content tab: files
```

Found it... looks like it changed to **plugin_gcodeviewer** at some point late last year.

jneilliii, 2021-12-13:
Yep, it was in 1.4.1 when it became extracted out as it's own plugin and the view model name changed. Listed in the release notes there.

https://github.com/OctoPrint/OctoPrint/releases/tag/1.4.1

closed 2021-12-13 by jneilliii

cp2004, 2021-12-13:
As it's a bundled plugin, you can just disable it entirely now in the plugin manager as well.

synman, 2021-12-13:
@cp2004 this is a really good idea for my use case.  Thanks for the tip!

synman, 2021-12-14:
@cp2004 & @jneilliii ironically the plugin itself is named gcodeviewer still.

```
plugins:
  _disabled:
  - printer_safety_check
  - fullscreen
  - plugin_gcodeviewer
  - gcodeviewer

```
I tried disabling "plugin_gcodeviewer" and it didn't work.  I then disabled it via Plugin Manager and the above is the result.