OctoPrint/OctoPrint #4346: [Request] Web UI Notification Plugin CLI

rudamaia, 2021-12-15:
### Describe the solution you'd like

I  have created a startup shell script to backup my octopi settings and push to github:
`#!/bin/sh
GIT=`which git`
REPO_DIR=/home/pi/.octoprint/data/backup
cd ${REPO_DIR}
rm * -rf
/home/pi/oprint/bin/octoprint plugins backup:backup --exclude 'uploads' --exclude 'timelapse'
unzip *.zip
rm *.zip
${GIT} add -A
${GIT} commit -m "Backup Sync"
${GIT} push`

I searched all over the documentation and google to find out a command that would simply display a notification in the Web UI "Backup Sync Done" but I believe the notification plugin does not have CLI.

So... that's it, would be nice to send the Web UI notifications trought shell script using CLI.

cp2004, 2021-12-15:
There isn't a generic notification plugin - are you thinking of the 'Printer Notifications' one? Which is specifically for when `//action: notification` is sent from the printer.

This isn't as easy as it sounds, maybe even impossible. With the CLI you are running commands in a separate process that doesn't have anything to do with the main server/web UI. How would the CLI process communicate with the server? Trying to go through the http API would require a key, and to know what port it is running on, for example. They can't be assumed.

rudamaia, 2021-12-15:
Yeah I assume it is not an easy job but I believe it would be a nice add-on for the ones using sh for different kinds of tasks. It woluld be nice to have "verbose info" about what your sh is doing. The same way I can start a backup I could send a custom notification to the UI.

The API would also be possible, the key part is the easy part, I could generate a App Key and use it in a Curl command. The thing is the API doesn't have this 'notification' functionality I think.
If it had a endpoint say '/api/system/commands/notification' to post a new custom notification would be great.

Just to be clear here, the notification I am proposing is the same simple notification that pops when you shutdown octoprint using the Web UI.

jneilliii, 2021-12-15:
This could be done with a plugin...