OctoPrint/OctoPrint #4348: XYZ Printing Da Vinci Mini W+ fails to connect after upgrading Octoprint

oilcan-productions, 2021-12-20:
### The problem

My XYZ Davinci Mini W+ used to work with Octoprint without an issue.
Since the last update the printer does not connect via serial anymore.
I rebuilt the system with the latest image. no dice.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPi 0.18.0 

### Printer model & used firmware incl. version

XYZ Print Da Vinci Mini W+ Latest FW version

### Browser and version of browser, operating system running browser

Microsoft Edge 96.0.1054.62, Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-logs.zip](https://github.com/OctoPrint/OctoPrint/files/7747736/octoprint-logs.zip)
[octoprint-systeminfo-20211220200950.zip](https://github.com/OctoPrint/OctoPrint/files/7747737/octoprint-systeminfo-20211220200950.zip)
`Starting dependency resolution...
packed_core.js?0d093a46:1257 ... dependency resolution done
packed_core.js?0d093a46:1280 Initial application setup done, connecting to server...
packed_core.js?0d093a46:1081 Connected to the server
packed_core.js?0d093a46:1087 Triggering connect on all view models
packed_core.js?0d093a46:1089 Server (re)connect processed
packed_core.js?0d093a46:1280 Finalizing application startup
packed_core.js?0d093a46:1261 Going to bind 38 view models...
packed_core.js?0d093a46:1273 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?0d093a46:781 Chrome Arc Fix disabled
packed_core.js?0d093a46:781 Chrome Arc Fix enabled
packed_core.js?0d093a46:1273 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_update since it does not exist
packed_core.js?0d093a46:1273 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_settings since it does not exist
packed_core.js?0d093a46:216 User octoprint logged in
packed_core.js?0d093a46:221 Login mechanism: http
packed_core.js?0d093a46:1278 ... binding done
packed_core.js?0d093a46:1278 Application startup complete`

oilcan-productions, 2021-12-20:
[serial (4).log](https://github.com/OctoPrint/OctoPrint/files/7747759/serial.4.log)
Addin Serial Log as well

cp2004, 2021-12-26:
You will probably find better help on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) to debug this issue. It looks like the printer is not replying to OctoPrint's connection attempts. It will be easier to troubleshoot on the forums/discord and then if the consensus is that it is still a bug in OctoPrint we can come back to this issue.

closed 2022-01-17 by foosel