OctoPrint/OctoPrint #4350: Getting 'pip command not found' error message when managing plugins

charlee, 2021-12-22:
### The problem

Getting 'pip command not found' error message when managing plugins. Inspecting the log file (`octoprint.log`) showed the following error when Octoprint was trying to update the plugins:

```
2021-12-22 15:46:32,178 - octoprint.plugin - ERROR - Error while calling plugin softwareupdate
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 224, in on_after_startup
    self._check_environment()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 362, in _check_environment
    and get_comparable_version(versions["pip"])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/version.py", line 170, in get_comparable_version
    version_string = normalize_version(version_string)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/version.py", line 211, in normalize_version
    if "-" in version:
TypeError: argument of type 'NoneType' is not iterable
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.7.2

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I found an workaround by doing the following steps. It would be great if this workaround could be added to [the FAQ](https://community.octoprint.org/t/i-just-installed-octoprint-1-3-12-octopi-0-17-0-and-the-plugin-manager-cant-find-pip-whats-up/15692/2).

I checked the `/home/pi/oprint/lib/python3/site-packages/pip` and found that the folder was empty.
Looks like the `pip` was not installed correctly.

Here is the solution:

1. Move to `/home/pi/oprint/bin`:

    cd /home/pi/oprint/bin

2. Download `get-pip.py`: 

    wget https://bootstrap.pypa.io/get-pip.py

3. Run `get-pip.py` to install pip. IMPORTANT: Make sure you are under `/home/pi/oprint/bin` and run python with `./python` so that pip is installed for OctoPrint. Running python with `python` or `python3` without prefixed by `./` will NOT work!

    ./python get-pip.py

4. Go to OctoPrint UI and reboot OctoPrint.
5. Go to OctoPrint Plugin Manager, click the Wrench icon, click "Save", then click "Save".

The problem should be solved now.

However I hope this issue could be solved in OctoPi.

cp2004, 2021-12-22:
Was this a fresh OctoPi install or one you have been using for any length of time?

charlee, 2021-12-22:
> Was this a fresh OctoPi install or one you have been using for any length of time?

This was a fresh OctoPi install, installed with Raspberry Pi Imager.
After I spotted this issue, I did a `sudo apt-get install python3-pip` before I realized that OctoPrint was using its own version of Python, but I don't think this would clear the `~/oprint/lib/python3/site-packages/pip` directory.

foosel, 2022-01-17:
I have never seen this before and according to your log pip IS there, OctoPrint just fails to figure out its version when trying to run software update. The OctoPrint venv on OctoPi certainly does ship with everything installed and ready to go, so whatever went wrong here is a *really* isolated case. Unable to reproduce this, the only thing I can do here is harden the code a bit further against an issue with an unparseable version number, which I guess might have been the case here. It's impossible to say though since there's no log on what OctoPrint did get back from pip here, so I'll also add some logging for that. That's about the only thing I can do here though, since, again, this should normally never happen and users should never have to manually install pip in the first place, something went wrong with this flash here and I have no clues as to what that was.

closed 2023-01-16 by foosel