OctoPrint/OctoPrint #4352: [Request] Timelapse pre delay

piotrenewicz, 2021-12-24:
### Is your feature request related to a problem? Please describe.

I wanted to use event hook octoprint.timelapse.capture.pre, to flash external lights, just for the picture to be taken.
The lights need a short delay to reach full brightness, and camera needs a further delay to adjust exposure.
Unfortunately the capture takes place immediately after calling the hook. And the resulting image is still black.

### Describe the solution you'd like

A configurable delay in milliseconds. Either exposed in webUI timelapse settings or hidden in config.yaml.
This delay would be applied after calling octoprint.timelapse.capture.pre, but before attempting to capture a frame.

### Describe alternatives you've considered

I've attempted to delay the frame capture, by blocking for half a second in the script executed by octoprint.timelapse.capture.pre event hook. But the frame capture didn't wait for my script to exit.

One obvious alternative would be Octolapse plugin, as it has this delay feature; 
But after trying Octolapse, i decided to stick with the octoprint.timelapse implementation, 
as it is much simpler to set up, and Octolapse is overkill in my situation.

### Additional context

Additionally, i wonder if the printer could continue receiving gcode commands in the meantime?
Would the default behavior be to temporarily interrupt the job, as the picture is taken?
And if so, then would it be feasible to make the delay and capture happen without interrupting the printing loop?

cp2004, 2021-12-26:
I'm surprised that a simple `time.sleep(<delay>)` did not work for you in that hook. They aren't run asynchronously, they are called one-by-one before the capture takes place.

https://github.com/OctoPrint/OctoPrint/blob/4b12b0e6f06c3abfb31b1840a0605e2de8e911d2/src/octoprint/timelapse.py#L733-L739

piotrenewicz, 2021-12-27:
Since making this Request, I've looked through the code, and found the section you just mentioned.
I've attempted to modify my local copy of this code, to include a sleep just after that for loop. 
This didn't stop the capture from happening immediately either. I thought i edited the wrong file, and my change wasn't executed.

During typing this comment i've realized what happened: I've set up the hook through the new webUI settings screen Event Manager.
Previously I didn't think it would be relevant, but it's actually the culprit. 
So looking back at that screen, it's obvious I'm not using hook capture.pre. I'm using Event CaptureStart instead.
And that mechanism is not blocking. This turned out to be user confusion.

So from here, one solution would be to delay after CaptureStart event.
Another would be for me to figure out how to specify an actual capture.pre hook.

It would be great to let people set blocking things through that new UI.
Would it be feasible to add the ability to set blocking hooks in that UI? not just event triggers.
If it is doable, then would you like me to open a new Request for that feature?

Waiting for a reply and will close this soon after.

foosel, 2022-01-17:
The new UI is merely a frontend for functionality that has been in OctoPrint for the better part of a decade - event hooks. Those are NOT the same as plugin hooks, and are asynchronous by nature. This can not be changed either, so using event hooks, it is impossible to create a blocking operation, you *need* a plugin for that. I will add a note for that to the editor however.

piotrenewicz, 2022-02-01:
Thanks for the info about hooks being a plugin only thing. That also wasn't obvious previously.
So i guess the best solution would be a plugin that exposes configuration of plugin hooks to the user in UI.
I've looked for such a plugin, but i can't find anything with this functionality.
I might attempt to create a plugin for this, at some point. (no time right now)

Anyway, as this is about plugin functionality, it's out of the scope of OctoPrint. 
Closing...

commit 77d57076 (2022-01-17): 💬 event mgr: Add note that event handlers are async

Closes #4352
closed 2022-02-01 by piotrenewicz