OctoPrint/OctoPrint #4353: Provide UI in settings to set the defaultExtrusionLength that defaults to 5mm

wilbur4321, 2021-12-27:
### Is your feature request related to a problem? Please describe.

There is no UI way to set `printer.defaultExtrusionLength` that was added in #513 (commit 6c5b32d92bc22765ed65339b75cf74702b7c604e). This is still used by the UI to set the default extrusion / retraction length in the control panel.

### Describe the solution you'd like

Provide way to set `printer.defaultExtrusionLength` through web ui.

jneilliii, 2021-12-27:
It does appear that this control was lost during a refactoring at some point. The input shown [here](https://github.com/OctoPrint/OctoPrint/commit/6c5b32d92bc22765ed65339b75cf74702b7c604e#diff-c310be9b0b1f74191ea31832e6c3c362ff4a3fec4671eeafdd8b1cba6ca7faacR144-R149) does not seem to be in the current maintenance branch.

jneilliii, 2021-12-27:
Should be added back to ths file https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/templates/snippets/settings/printerprofiles/profileEditorExtruder.jinja2 I believe.

jneilliii, 2021-12-27:
I guess the question here though is did the logic change somewhere else, which was the reason it was removed (ie updating on the Control Panel updates the setting)?

wilbur4321, 2021-12-27:
I saw this as well. The API code path still appears to work end-to-end, as I modified the settings in the browser debugger in the callback on the `save` button in the UI.

It seems like it should belong to a per-printer-profile config (if not per extruder?), which makes me wonder if it got lost then.

Additionally, the setting seems oddly placed as the only thing under `printerParameters:` (at the root) in the config yaml file.

jneilliii, 2021-12-27:
yeah, I think something is broken here, because I would suspect that it would be included as part of the printer's profile and not a root setting, but I could be wrong. I was just looking on my dev machine at the files stored in .octoprint/printerProfiles nd that extrusionlength was not in any of the ones I have, which is where I would suspect it to be stored.

wilbur4321, 2021-12-27:
> Should be added back to ths file https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/templates/snippets/settings/printerprofiles/profileEditorExtruder.jinja2 I believe.

The actual setting is definitely _not_ stored in the printer profile nor is profile specific, however. This might need a simple fix to put it in the UI somewhere, then a deeper change to make it per-profile at a minimum, I don't have a multi-extruder printer, so I don't know if the UI/api does overrides on a per-extruder basis usually, also?

jneilliii, 2021-12-27:
So there is a the default printerProfiles section of config.yaml, so it probably is right there, but having extruder specific ones does make sense.

https://docs.octoprint.org/en/master/configuration/config_yaml.html#printer-profiles

wilbur4321, 2021-12-27:
> yeah, I think something is broken here, because I would suspect that it would be included as part of the printer's profile and not a root setting, but I could be wrong. I was just looking on my dev machine at the files stored in .octoprint/printerProfiles and that extrusionlength was not in any of the ones I have, which is where I would suspect it to be stored.

It got stored into ~/.octoprint/config.yaml on my klipper machine when I set it via API:

```yaml
printerParameters:
  defaultExtrusionLength: 50
```

closed 2022-05-17 by cp2004