OctoPrint/OctoPrint #4354: New PRUSA Firmware screen display bug

martinglass, 2021-12-27:
Upgraded my Prusa MK3S+ to firmware v3.10 (prusa3d_fw_MK3S_3_10_0_4481.hex)

When starting a print from file in Octoprint, the LCD display on the Prusa shows white blocks instead of data, on many of the lines (not all).

I've attached a photo to show the problem (the data that isn't white blocks flashes at about 1Hz
[image]

I can send a logifle, if you can give me an email address.

cp2004, 2021-12-27:
If it worked fine on the old firmware but not on the new one, why is this an OctoPrint issue?

martinglass, 2021-12-27:
> If it worked fine on the old firmware but not on the new one, why is this an OctoPrint issue?

You're right. I've checked printing from SD Card and it happens that way too. So, it's not an OctoPrint issue.

Thanks.

I'll close this issue.

martinglass, 2021-12-27:
I've checked printing from SD Card and it happens that way too. So, it's not an OctoPrint issue.

closed 2021-12-27 by martinglass